sappelhoff/pyprep #2: Implementing the entire PREP pipeline

Nick3151, 2019-11-13:
Hi! We'd like to implement the entire PREP pipeline, which will include:

1. Bug fixing of Noisydata class
2. Feature enhancement by adding filtering, line noise removal and robust referencing.
3. Integrating all the steps into one function

sappelhoff, 2019-11-13:
Hi @Nick3151 - that sounds great to me. Who is "we"? How do you want to work on this?

Nick3151, 2019-11-13:
And maybe we could get pyprep merged into MNE in the future.

sappelhoff, 2019-11-14:
sounds very good, let's continue the discussion in https://github.com/mne-tools/mne-python/issues/7049

closed 2019-11-14 by sappelhoff

adam2392, 2019-11-15:
@sappelhoff How do you feel that these folks, @Nick3151 and two others work on a PR into the forked version of this in https://github.com/NeuroDataDesign/pyprep and then we submit a PR here?

Or should we directly try to address the fixes mentioned here and in the discussion in mne-tools/mne-python#7049 with a direct PR in the main repo?

Let us know your thoughts.

sappelhoff, 2019-11-15:
I think your first suggestion works well. That way you can work independently of any GitHub rights you may need to move forward.

It'd be great to tag me occasionally when there is new stuff you integrate, ... that way I won't be too overwhelmed once you make a PR from your fork to this repo :-)