sappelhoff/pyprep #5: Implementation of Full PREP and Python3 compatability

adam2392, 2019-12-01:
Hi @sappelhoff, @Nick3151 and @AamnaLawrence here are working on migrating the full functionality of PREP into this forked version of pyprep. I'm overseeing the project and will help wherever they are lacking some skillsets. 

This will involve a variety of factors:

1. Scikit-learn "fit" style pipeling
2. Turning things into classes where it is relevant.
3. Adding additional unit/integration tests for the above
4. Adding python3 into travis.ci 

I'm just making an issue here, so you can comment on any of these above things and guide dev. For PRs on a forked version, you can check out dev at anytime at: #4 #3 and https://github.com/NeuroDataDesign/pyprep/pulls.

Lmk your thoughts!

sappelhoff, 2019-12-02:
Thanks a lot @adam2392 :-)

neerajww, 2020-02-06:
Are there any updates on this?

sappelhoff, 2020-02-06:
This was closed with #6

closed 2020-02-06 by sappelhoff

sappelhoff, 2020-02-10:
@neerajww I streamlined that inclusion a bit and updated the docs. You can check out the bottom part of the examples page:https://pyprep.readthedocs.io/en/latest/examples.html

it now links to: https://nbviewer.jupyter.org/github/sappelhoff/pyprep/blob/master/examples/prep_demo.ipynb

neerajww, 2020-02-15:
@sappelhoff Thank you very much. I am eager to try this out on my dataset. As I installed, I am faced the following issues which hints something is missing in my installation. Can you help me on this.
``
Step 1: pip install pyprep
This threw error:
ERROR: Cannot uninstall 'psutil'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
Step 2: sudo pip install --ignore-installed pyprep
This gave the message:
Successfully installed mne-0.19.2 numpy-1.18.1 pandas-1.0.1 patsy-0.5.1 psutil-5.6.7 pyprep-0.2.3 python-dateutil-2.8.1 pytz-2019.3 scipy-1.4.1 six-1.14.0 statsmodels-0.11.0
Step 3: I typed the following in Jupyter Notebook:
import pyprep
dir(pyprep)
This gives the following output:
['__builtins__',
 '__cached__',
 '__doc__',
 '__file__',
 '__loader__',
 '__name__',
 '__package__',
 '__path__',
 '__spec__']

Step 4: I guess something is missing in the installation because
from pyprep.prep_pipeline import PrepPipeline
This gives the following output:
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-6-c8198614311d> in <module>
----> 1 from pyprep.prep_pipeline import PrepPipeline

ModuleNotFoundError: No module named 'pyprep.prep_pipeline'
``

sappelhoff, 2020-02-15:
Hi @neerajww - I have updated the installation instructions. please check out: https://github.com/sappelhoff/pyprep/blob/master/README.md#installation

regarding using pyprep, please check out the examples: https://pyprep.readthedocs.io/en/latest/examples.html

You tried to use `pyprep.prep_pipeline` --> this module is only available in the **development** version of the package. That is, it has not yet been released. I have made a comment to the example covering `pyprep.prep_pipeline` that you need to install the **development** version in order to use it. See here: https://github.com/sappelhoff/pyprep/blob/master/examples/prep_demo.ipynb

By using `pip install pyprep` you can only use the features included in the **stable** version that has been released on pypi.

neerajww, 2020-02-15:
Thank you @sappelhoff . It seems to take the raw MNE object as input. Is this necessary requirement for bad channel detection? Can it not work on epoch object as well.

sappelhoff, 2020-02-17:
Yes, pyprep works only on raw data. Not sure if it could be extended to work on epochs, feel free to look into that and let me know.