sappelhoff/pyprep #7: ValueError: Array Must Not Contain Infs or NaNs

marcoistasy, 2020-03-20:
I seem to be having an issue when running this pipeline. Below is the traceback:

```
RuntimeWarning: invalid value encountered in greater
  mask = (size > 0)
Traceback (most recent call last):
  File "/Users/opt/anaconda3/envs/mytfenv/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3319, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-4-997c912b5fa5>", line 23, in <module>
    prep_pipeline.fit()
  File "/Users/Documents/preprocessing/prep_pipeline.py", line 80, in fit
    reference.perform_reference()
  File "/Users/Documents/preprocessing/reference.py", line 66, in perform_reference
    self.robust_reference()
  File "/Users/Documents/preprocessing/reference.py", line 144, in robust_reference
    noisy_detector.find_all_bad_channels(ransac=self.ransac)
  File "/Users/Documents/preprocessing/find_noisy_channels.py", line 140, in find_all_bad_channels
    self.find_bad_by_ransac()
  File "/Users/Documents/preprocessing/find_noisy_channels.py", line 476, in find_bad_by_ransac
    n_samples=n_samples,
  File "/Users/Documents/preprocessing/find_noisy_channels.py", line 581, in run_ransac
    chn_pos, chn_pos_good, good_chn_labs, n_pred_predicted, data
  File "/Users/Documents/preprocessing/find_noisy_channels.py", line 632, in get_ransac_pred
    interpol_mat = _make_interpolation_matrix(reconstructed_positions, chn_pos)
  File "/Users/opt/anaconda3/envs/mytfenv/lib/python3.7/site-packages/mne/channels/interpolation.py", line 85, in _make_interpolation_matrix
    C_inv = linalg.pinv(C)
  File "/Users/opt/anaconda3/envs/mytfenv/lib/python3.7/site-packages/scipy/linalg/basic.py", line 1304, in pinv
    a = _asarray_validated(a, check_finite=check_finite)
  File "/Users/opt/anaconda3/envs/mytfenv/lib/python3.7/site-packages/scipy/_lib/_util.py", line 246, in _asarray_validated
    a = toarray(a)
  File "/Users/opt/anaconda3/envs/mytfenv/lib/python3.7/site-packages/numpy/lib/function_base.py", line 499, in asarray_chkfinite
    "array must not contain infs or NaNs")
ValueError: array must not contain infs or NaNs
```

I understand that ValueError: array must not contain infs or NaNs means that there are some non numerical values in my array. However, when I check whether there are in fact NaNs or infs values in my array using this answer I get False as the result (i.e there are no infs or NaNs). Is there anyway to fix this issue?

sappelhoff, 2020-03-20:
This error seems to appear in the new code. Maybe @adam2392 @AamnaLawrence or @Nick3151 can help?

yjmantilla, 2020-03-20:
What array are you referring to when you say you checked the array? Do you mean the actual eeg data (as in raw._data[channels,time]) or do you mean the array which has the positions of the electrodes?

From what I understand the error is in calculating the interpolation matrix, it seems the function pinv is encountering some NaNs or Infs , the 'a' array it refers to in the final part of the traceback is calculated from the positions of the electrodes. 

So if I were you I would check all of the positions of the electrodes are available.

marcoistasy, 2020-03-20:
> What array are you referring to when you say you checked the array? Do you mean the actual eeg data (as in raw._data[channels,time]) or do you mean the array which has the positions of the electrodes?
> 
> From what I understand the error is in calculating the interpolation matrix, it seems the function pinv is encountering some NaNs or Infs , the 'a' array it refers to in the final part of the traceback is calculated from the positions of the electrodes.
> 
> So if I were you I would check if all of the positions of the electrodes are available.
> 
> If all are available then maybe the problem is when applying the transformations of that array that in your particular case is producing some NaNs or Infs. In particular the interpolation code does:
> 
> ```
> 1) normalize sensor positions to sphere
> 2) calculate cosine angles between source positions
> 3) Calculate spherical spline g function between points on a sphere.
> 4) ... and more stuff im not sure about
> ```
> 
> So the point is that along that way maybe some Infs and NaNs appear prior to the calculation of the inverse by the pinv function.
> 
> *** This is my interpretation of the error, so I may be mistaken ***

Hi! Thanks so much for your reply. I am not interested in the position of the electrodes as this doesn't concern my current project, so I am assuming that the code is referring to the actual EEG data. 

In fact, I am using a customised version of this code which does not include a `montage`.  Is the `interpolation matrix` which channels or readings? 

I apolgise if this does not make much sense - I am quite new to eeg analysis. Would it make more sense to share my code with you?

yjmantilla, 2020-03-20:
> Hi! Thanks so much for your reply. I am not interested in the position of the electrodes as this doesn't concern my current project, so I am assuming that the code is referring to the actual EEG [...]

If you want. In any case from what I know (Im no expert) to be able to do interpolation of channels prep will require a montage because the interpolation is done on the basis of the locations (is a spatial interpolation) ie electrodes near a given electrode contribute more for the interpolation. 

If there is no montage then there will be no spatial criteria to use for interpolation. If i were you I would check the source of your eeg data for information of where the electrodes were located, if that information is not available I would be weird to try to interpolate at all. Maybe you dont have exact locations but the montage could have followed an standard montage so in that case your montage is a subset of the standard one. In that case just identify the standard montage that best fits your case. MNE will inform that the channels are a subset of the montage and thats it, it will work.

In the context of the prep from what i know interpolation is done for 2 reasons:
- Estimate bad channels using the RANSAC criteria
- Fabricate data for bad eeg channels (interpolate bad channels that is).

If you dont have any location data its problematic because the robust referencing algorithm involves interpolation of bad channels. 

Now that you say that the code is customized it makes sense for this to maybe happen, I mean I think internally pyprep verifies that location data is available but if that part of the code was deleted I would make sense for this error to appear.

marcoistasy, 2020-03-20:
> > > What array are you referring to when you say you checked the array? Do you mean the actual eeg data (as in raw._data[channels,time]) or do you mean the array which has the positions of the electrodes?
> > > From what I understand the error is in calculating the interpolation matrix, it seems the function pinv is encountering some NaNs or Infs , the 'a' array it refers to in the final part of the traceback is calculated from the positions of the electrodes.
> > > So if I were you I would check if all of the positions of the electrodes are available.
> > > If all are available then maybe the problem is when applying the transformations of that array that in your particular case is producing some NaNs or Infs. In particular the interpolation code does:
> > > ```
> > > 1) normalize sensor positions to sphere
> > > 2) calculate cosine angles between source positions
> > > 3) Calculate spherical spline g function between points on a sphere.
> > > 4) ... and more stuff im not sure about
> > > ```
> > > 
> > > 
> > > So the point is that along that way maybe some Infs and NaNs appear prior to the calculation of the inverse by the pinv function.
> > > *** This is my interpretation of the error, so I may be mistaken ***
> > 
> > 
> > Hi! Thanks so much for your reply. I am not interested in the position of the electrodes as this doesn't concern my current project, so I am assuming that the code is referring to the actual EEG data.
> > In fact, I am using a customised version of this code which does not include a `montage`. Is the `interpolation matrix` which channels or readings?
> > I apolgise if this does not make much sense - I am quite new to eeg analysis. Would it make more sense to share my code with you?
> 
> If you want. In any case from what I know (Im no expert) to be able to do interpolation of channels prep will require a montage because the interpolation is done on the basis of the locations (is a spatial interpolation) ie electrodes near a given electrode contribute more for the interpolation.
> 
> If there is no montage then there will be no spatial criteria to use for interpolation. If i were you I would check the source of your eeg data for information of where the electrodes were located, if that information is not available it would be weird to try to interpolate at all. Maybe you dont have exact locations but the montage could have followed an standard montage so in that case your montage is a subset of the standard one. In that case just identify the standard montage that best fits your case. MNE will inform that the channels are a subset of the montage and thats it, it will work.
> 
> In the context of the prep from what i know interpolation is done for 2 reasons:
> 
> * Estimate bad channels using the RANSAC criteria
> * Fabricate data for bad eeg channels (interpolate bad channels that is).
> 
> If you dont have any location data its problematic because the robust referencing algorithm involves interpolation of bad channels.
> 
> Now that you say that the code is customized it makes sense for this to maybe happen, I mean I think internally pyprep verifies that location data is available but if that part of the code was deleted It would make sense for the shown error to appear.

thanks so much! that's what i was getting at too and I realised that is the case (i.e can't interpolate because there is no data). I am trying to remove the interpolation features from the code -  could you point me to the function in charge of interpolation so that I can comment it out?

yjmantilla, 2020-03-20:
That would be tricky, from the top of my mind I can say:

- put ransac in false (this will take care of the make_interpolation_matrix part)

- in perform reference you would need to comment anything that interpolates

the functions that interpolates is .interpolate_bads()...

THERE IS A BIG PROBLEM WITH THIS: Interpolation is an essential part of the PREP idea of robust referencing, im not sure what would be the effect of it if you just comment those parts. Basically the prep idea is to iteratively calculate a reference by interpolating bad channels. 

To change it would require really thinking through what are you really doing and seeing if it really makes sense.PREP does:

- Filter line noise by while first doing a detrend --> ok, doesnt need interpolation
- Calculate robust reference --> needs interpolation ... what to do here then?

From the top of my mind:
You could just calculate a reference by using the average of non bad channels (which were detected using every criteria except ransac) and not interpolate anything. Exclude from further analysis those bad channels.

Point is , you would need to think about it to make sure it makes sense.

marcoistasy, 2020-03-20:
> That would be tricky, from the top of my mind I can say:
>
> * put ransac in false (this will take care of the make_interpolation_matrix part)
> * in perform reference you would need to comment anything [...]

Thanks so much for your fast reply! Would you be able to help me with my code, by any chance? I removed all `ransac` methods from `noisyChannels` but am getting stumped while removing `interpolation` from `Reference`.

Alternatively, what I was thinking to do was simply to find the bad channels and remove them while forgoing the process of referencing. Please let me know your thoughts!

yjmantilla, 2020-03-20:
> Thanks so much for your fast reply! Would you be able to help me with my code, by any chance? I removed all `ransac` methods from `noisyChannels` but am getting stumped while removing [...]

I think that last idea will work, basically I think is pretty much the best you can do (that i know of) without having to interpolate. In this case the reference would just be

np.nanmedian(raw.get_data(picks=self.reference_channels), axis=0) * 1e6

where reference_channels are the good channels.

If you are keep getting stumped at it maybe we could do a google meet while sharing your screen and with the questions you have of the code.

marcoistasy, 2020-03-20:
a google meet would be much appreciated! my email is listed on my github profile - would you be able to reach out and we can set it up through email? Thank you!

(I'll mark this as closed!)

closed 2020-03-20 by marcoistasy

yjmantilla, 2020-03-20:
sure, I sent you an email

sappelhoff, 2020-03-21:
If you find that the issue has something to do with the codebase, or that there could be improvements to make the codebase more robust, please make a PR! :-)