sappelhoff/pyprep #13: MAINT. release v0.3

sappelhoff, 2020-04-04:
once #9 and some other fixes that @yjmantilla wanted to do are merged, we can release a new version of `pyprep`.

yjmantilla, 2020-04-17:
Also thinking on usability (specifically #16) : when we do the refactor a bit of renaming should be helpful (ie prepPipeline.raw should be renamed so it is associated with the actual processed eeg , not solely 'raw')

sappelhoff, 2020-06-04:
I released `0.3.0` today and did several commits for maintenance, fixing tests, improving docs, etc.

there is a little more automation in place now which will make future maintenance of this package easier

I also added a big disclaimer that this software is in ALPHA phase (use at your own risk) and that we are searching for contributors.

@yjmantilla you should also have a new look at https://github.com/sappelhoff/pyprep#installation and update your development environment --> I configured pre-commit hooks for this project, which will hopefully make the integration with "black" a bit more smooth in the future.

closed 2020-06-04 by sappelhoff

yjmantilla, 2020-06-04:
Awesome!  @sappelhoff
  
Sorry I haven't contributed much from the start of covid. Paradoxically college classes and work have been more stressful so I have not been able to find the time. I already finished 2 courses this week but still got left the hardest ones. I expect to have much more free time in about a month, when I have already got out of the main evaluations. Until then I may contribute a bit but can't really promise anything.

I checked https://github.com/sappelhoff/pyprep#installation  and it is really nice to have the automated black commits. Now, apart from the the good old rebase, is there anything needed to do new contributions?

sappelhoff, 2020-06-05:
Ah sorry to hear that it's been a stressfull time for you ... no worries -> I just wanted to keep you up to date.

> Now, apart from the the good old rebase, is there anything needed to do new contributions?

nope, just doing `pip install -r requirements-dev.txt` again and then `pre-commit install` from the project root.