sappelhoff/pyprep #16: Question regarding pyprep

SebastianSpeer, 2020-04-06:
Dear Pyprep team,

I'm currently using the pyprep pipeline and it has been quite helpful so far. However, I am confused  about which of the the attributes in the prep output is the final preprocessed data? Is it EEG_clean or is it EEG_new? In addition, how can the cleaned data be integrated back into the mne data structure?

Thanks for your help,

Sebastian

sappelhoff, 2020-04-06:
@AamnaLawrence, @adam2392, @Nick3151can one of you help with this? :-)

adam2392, 2020-04-06:
@SebastianSpeer I believe it is just `prep.EEG` (?)

See: https://github.com/sappelhoff/pyprep/blob/master/examples/run_full_prep.py

I agree this is a bit... not intuitive. @sappelhoff sorry I'm a bit swamped w/ the COVID situation rn, but I can definitely tend to this perhaps in a week or two? I think some refactoring and cleaning is necessary. I see that some of the usability issues are being hashed out in some of the other issues.

sappelhoff, 2020-04-06:
> I think some refactoring and cleaning is necessary. I see that some of the usability issues are being hashed out in some of the other issues

indeed! But it's a good sign that we and others are noticing this, because it shows that there's an interest for the package :-)

I did some housekeeping work in the last two weeks, but did not really touch the "full PREP" code yet. Would be much appreciated if you could find the time in a week or two @adam2392. See also #9 where @yjmantilla is starting to fix some bugs.

> sorry I'm a bit swamped w/ the COVID situation rn

no worries, stay safe :-)

@SebasianSpeer, if in the meantime you figure out how to improve the documentation by looking at the source code, feel free to send over a pull request as well!

yjmantilla, 2020-04-06:
From what I know the cleaned (as in no line noise and robust reference) is in the prepPipeline.raw mne object. It is counter intuitive indeed. To support this just look at perform_reference method in reference.py, at the end it is said:

`# MNE Raw object after interpolation`
`self.raw._data = self.EEG * 1e-6`

In that sense the "cleaned" is already in an mne structure.

Nick3151, 2020-04-06:
@SebastianSpeer Yeah, you can use [EEG_final = prep.raw.get_data()*1e6](https://github.com/sappelhoff/pyprep/blob/master/examples/run_full_prep.py#L204) to get the final output EEG. Bsically PREP detects a group of bad channels and then interpolate them. So the final EEG is actually the EEG after interpolation.

Nick3151, 2020-04-06:
For the sake of validating the outputs against the Matlab version, some variables are kind of confusing in the code. I'm sorry for that and will make some contributions to make in more the functions more user-friendly. Please point it out if you find some other issues, thanks!

SebastianSpeer, 2020-04-06:
> From what I know the cleaned (as in no line noise and robust reference) is in the prepPipeline.raw mne object. It is counter intuitive indeed. To support this just look at perform_reference method in reference.py, at the end it is said:
> 
> `# MNE Raw object after interpolation`
> `self.raw._data = self.EEG * 1e-6`
> 
> Here self would be the reference class.
> 
> Then in prepPipeline is it made:
> `self.raw = reference.raw`
> 
> Here self is the prepPipeline class
> 
> In that sense the "cleaned" is already in an mne structure.

Okay great, thanks! So if I understood correctly I could I can use prep.raw (indeed counter-intuitive) for further processing steps in MNE (e.g. eye blink removal, epoching etc)? Is the multiplication with 1e6 important?

sappelhoff, 2020-04-06:
> Is the multiplication with 1e6 important?

The multiplication of the data with 1e6 is only important if you want your data to be in `µV`, because MNE-Python represents all data internally as `V`, see: https://mne.tools/stable/overview/implementation.html?highlight=units#id20

yjmantilla, 2020-04-06:
TLDR: Just use prep.raw after running fit to have the preprocessed signal as an mne object.

`# MNE Raw object after interpolation
self.raw._data = self.EEG * 1e-6 `

Yes exactly. The multiplication by 1e-6 is because internally we tried to stay close to matlab which had the signal in microvolts but in mne signals are in volts. 

When @Nick3151  said

> @SebastianSpeer Yeah, you can use [EEG_final = prep.raw.get_data()*1e6](https://github.com/sappelhoff/pyprep/blob/master/examples/run_full_prep.py#L204) to get the final output EEG. Bsically PREP [...]

He multiplied by 1e6 to get the signal in microvolts as matlab's way since prep.raw.get_data() returns the signal in volts.

Just conversion stuff.

yjmantilla, 2020-04-17:
@SebastianSpeer  I think we can close this now right?

SebastianSpeer, 2020-04-17:
> @SebastianSpeer I think we can close this now right?

Yes, thanks for your help

closed 2020-04-17 by sappelhoff

SebastianSpeer, 2021-11-03:
Hi pyprep team,

Is it still the case that: EEG_final = prep.raw.get_data()*1e6 . Is the way to get the final preprocessed data or has the naming been changed to something more intuitive since?

a-hurst, 2021-11-03:
@SebastianSpeer, you can get an MNE Raw object directly from PyPREP by accessing the `PrepPipeline`'s `raw` attribute. After you've run `fit()` on the pipeline it'll contain the processed data, along with any non-EEG channels reappended. Let me know if that works!

SebastianSpeer, 2021-11-04:
@a-hurst, yes that works. Great. Does the data still need to be multiplied with:

> 1e6

a-hurst, 2021-11-04:
> @a-hurst, yes that works. Great. Does the data still need to be multiplied with:
> 
> > 1e6

Only if you want to convert it to microvolts instead of MNE's default internal unit of Volts! Otherwise, just treat it like you would any other MNE Raw object.