sappelhoff/pyprep #17: RANSAC consumes too much RAM

yjmantilla, 2020-04-06:
I have been testing pyprep against our datasets (GRUNECO, University of Antioquia, Colombia). For example I have 58 channels @ 1000Hz for about 5 minutes. This will need like 10 gb of RAM awarded from the OS to python. This is quite a lot for personal computers.

I have two possible proposals to solve this:
- Run RANSAC in a channel wise manner. 
- Automatically identify the resample needed for the data to fit memory 

I like more the first proposal and @sappelhoff  agrees with this approach. This was discussed in #9 .

Once #9 detrend_correction is merged into master I will upload this code so it may be reviewed and tested.

sappelhoff, 2020-04-07:
looking forward to a PR, I hope this would not slow down RANSAC too much :+1:

yjmantilla, 2020-04-07:
Im kind of academically compromised right now. I could do a pull request with the code as is, the problem is that is has no explanation. Or maybe by the time @adam2392  is available I will be more available too.

yjmantilla, 2020-04-08:
Regarding #8 and #9 this is a comment to remind me that I forgot to include another do_detrend = False in another call to noisyChannels in the robust reference function. I will fix this in the channel-wise ransac PR.

PR #32 (2020-07-06): lesser ram ransac
# PR Description

Adding a lesser hungry version of the ransac albeit slower since it is done channel by channel. As of now it will try to execute the fast ransac and if that fails will try to do the channel-wise ransac. It is about 5 to 6 times slower than the ram hungry version.

Would close #17 and close #19 also closes #28 

Edit: Optimized, now ransac will automatically find the largest chunk of channels it can take at a time without running out of ram.
closed 2020-07-06 by sappelhoff