sappelhoff/pyprep #19: Slight inconsistency in perform_reference

yjmantilla, 2020-04-06:
In reference.py we have the method perform_reference() which states the following:
```
        # Phase 1: Estimate the true signal mean with robust referencing
        self.robust_reference()
        if self.noisy_channels["bad_all"]:
            self.raw.info["bads"] = self.noisy_channels["bad_all"]
            self.raw.interpolate_bads() <----------------------------------NOTICE THIS
        self.reference_signal = (
            np.nanmean(self.raw.get_data(picks=self.reference_channels), axis=0) * 1e6
        )
        rereferenced_index = [
            self.ch_names_eeg.index(ch) for ch in self.rereferenced_channels
        ]
        self.EEG = self.remove_reference(
            self.EEG, self.reference_signal, rereferenced_index
        )
```

Now in the original Matlab code this is equivalent to doRobustPost() inside performReference.m:

```
    function [] = doRobustPost()
        % Robust reference with interpolation afterwards
        referenceOut = robustReference(signal, referenceOut);
        noisy = referenceOut.badChannels.all;
        if isempty(noisy)   %No noisy channels -- ordinary ref
            referenceOut.referenceSignal = ...
                nanmean(signal.data(referenceOut.referenceChannels, :), 1);
        else
            sourceChannels = setdiff(referenceOut.evaluationChannels, noisy);
            signalNew = interpolateChannels(signal, noisy, sourceChannels); <----- NOTICE THIS
            referenceOut.referenceSignal = ...
                nanmean(signalNew.data(referenceOut.referenceChannels, :), 1);
            clear signalNew;
        end
        signal = removeReference(signal, referenceOut.referenceSignal, ...
            referenceOut.rereferencedChannels);
        referenceOut.noisyStatistics  = ...
            findNoisyChannels(removeTrend(signal, referenceOut), referenceOut);
        
```

So the basic difference is that robustReference produce a list of bad channels and this are interpolated in a dummy variable then averaged to produce a robust reference. In pyprep the interpolation is done on the raw object directly, not on a dummy copy of it.

I'm not so sure if this affects greatly or not but I would like your opinion on this.

If we wanted to make the code more alike the original one we would:

```
        # Phase 1: Estimate the true signal mean with robust referencing
        self.robust_reference()
        if self.noisy_channels["bad_all"]:
            dummy = self.raw.copy()
            dummy.info["bads"] = self.noisy_channels["bad_all"]
            dummy.interpolate_bads()
            self.reference_signal = (
                np.nanmean(dummy.get_data(picks=self.reference_channels), axis=0) * 1e6
            )
            del dummy
        else:
            self.reference_signal = (
                np.nanmean(self.raw.get_data(picks=self.reference_channels), axis=0)
                * 1e6
            )
        rereferenced_index = [
            self.ch_names_eeg.index(ch) for ch in self.rereferenced_channels
        ]
        self.EEG = self.remove_reference(
            self.EEG, self.reference_signal, rereferenced_index
        )

```

sappelhoff, 2020-04-07:
> I'm not so sure if this affects greatly or not but I would like your opinion on this.

It seems to me that the only difference between the current pyprep implementation and the new implementation that you propose is:

1. in the current one, `self.raw` has all channels interpolated, while in the new implementation, `self.raw` is unchanged
1. the new implementation would need more memory, because we are making the `dummy` copy

so it is important what happens later in the code with `self.raw` to know whether your new proposed implementation would have any effect at all :thinking:

yjmantilla, 2020-04-07:
Basically that self.raw will become the final signal of the prepPipeline. Overall after the first rereference (phase 1 in code, true signal mean): 
- it calculates bad channels again
- interpolates them (will be interpolated_channels)
- calculates yet a new reference and removes it again
- find bad channels again (will be still_noisy_channels).

In that regard, in the original prep the final signal is only interpolated once. In the 1st rereference it interpolates on a **copy** just to calculate the **true signal mean** using it.

Here is a copy of what follows exactly in our code:

```
     # Phase 2: Find the bad channels and interpolate
        self.raw._data = self.EEG * 1e-6
        noisy_detector = NoisyChannels(self.raw)
        noisy_detector.find_all_bads(ransac=self.ransac)

        # Record Noisy channels and EEG before interpolation
        self.bad_before_interpolation = noisy_detector.get_bads(verbose=True)
        self.EEG_before_interpolation = self.EEG.copy()

        bad_channels = _union(self.bad_before_interpolation, self.unusable_channels)
        self.raw.info["bads"] = bad_channels
        self.raw.interpolate_bads()
        reference_correct = (
            np.nanmean(self.raw.get_data(picks=self.reference_channels), axis=0) * 1e6
        )
        self.EEG = self.raw.get_data() * 1e6
        self.EEG = self.remove_reference(
            self.EEG, reference_correct, rereferenced_index
        )
        # reference signal after interpolation
        self.reference_signal_new = self.reference_signal + reference_correct
        # MNE Raw object after interpolation
        self.raw._data = self.EEG * 1e-6

        # Still noisy channels after interpolation
        self.interpolated_channels = bad_channels
        noisy_detector = NoisyChannels(self.raw)
        noisy_detector.find_all_bads(ransac=self.ransac)
        self.still_noisy_channels = noisy_detector.get_bads()
        self.raw.info["bads"] = self.still_noisy_channels
        return self
```

yjmantilla, 2020-04-07:
> 2. the new implementation would need more memory, because we are making the `dummy` copy

Yeah but hypothetically if it is able to handle ransac (even in the channel wise implementation) then this will be negligible. After all it would delete this new dummy signal once it calculates the true mean.

sappelhoff, 2020-04-08:
> In that regard, in the original prep the final signal is only interpolated once. In the 1st rereference it interpolates on a copy just to calculate the true signal mean using it.

okay, I see ... so it's really a deviation from MATLAB prep ... don't know if there was any reason for it to be implemented that way :-/

in that case I agree with you and we should change the behavior.

What makes me generally uncomfortable, that we have only few tests implemented in pyprep right now ... so every time we change something, there is a small chance that we break something without noticing :-/ but that's a separate issue.

PR #32 (2020-07-06): lesser ram ransac
# PR Description

Adding a lesser hungry version of the ransac albeit slower since it is done channel by channel. As of now it will try to execute the fast ransac and if that fails will try to do the channel-wise ransac. It is about 5 to 6 times slower than the ram hungry version.

Would close #17 and close #19 also closes #28 

Edit: Optimized, now ransac will automatically find the largest chunk of channels it can take at a time without running out of ram.
commit 6bb931c0 (2020-07-06): lesser ram ransac (#32)

* lesser ram ransac

* fixed bare except

* channel wise ransac test added

* formatting fixes

* minor fixes

* Update docs/whats_new.rst

* Update examples/run_ransac.py

* Update examples/run_ransac.py

* fixed code smell

* reduced code redundancy, fixed #19 and another do_detrend problem

* format fixing

* more format fixing

* deleted one if in reference.py

* optimization by doing chunks of m channels that fit in ram

* format fixes

* added comment about frac_bad in correlation criteria

* added test for too little ram for find_noisy_channels

* format fixes

 [... cut]
closed 2020-07-06 by sappelhoff