sappelhoff/pyprep #20: Be able to see the result of each criteria in the prepPipeline object.

yjmantilla, 2020-04-06:
Currently the prepPipeline object has access to the bad channels before and after interpolation but it doesn't identify which criteria marked which channel. In the original PREP this info is in the EEG.etc.referenceOut.noisyStatisticsBeforeInterpolation and afterInterpolation. 

I understand that maybe for most users this info won't be needed but for some power users and contributors may find this useful for catching bugs, inconsistencies and in general test the criterias not isolated from each other and in the context of the whole pipeline.

I approached this by making the noisy detectors attributes of the reference class as follows:

```
        # Phase 2: Find the bad channels and interpolate
        self.raw._data = self.EEG * 1e-6
        self.noisy_detector_before_interpolation = NoisyChannels(self.raw)
        self.noisy_detector_before_interpolation.find_all_bads(ransac=self.ransac)

        # Record Noisy channels and EEG before interpolation
        self.bad_before_interpolation = self.noisy_detector_before_interpolation.get_bads(verbose=True)
        self.EEG_before_interpolation = self.EEG.copy()

.................

        # Still noisy channels after interpolation
        self.interpolated_channels = bad_channels
        self.noisy_detector_after_interpolation = NoisyChannels(self.raw)
        self.noisy_detector_after_interpolation.find_all_bads(ransac=self.ransac)
        self.still_noisy_channels = self.noisy_detector_after_interpolation.get_bads()
```

And then in prepPipeline object do:

```
        self.noisy_detector_before_interpolation = \
        { \
        'bad_by_nan' : reference.noisy_detector_before_interpolation.bad_by_nan,
        'bad_by_flat' : reference.noisy_detector_before_interpolation.bad_by_flat,
        'bad_by_deviation' : reference.noisy_detector_before_interpolation.bad_by_deviation,
        'bad_by_hf_noise' : reference.noisy_detector_before_interpolation.bad_by_hf_noise,
        'bad_by_correlation' : reference.noisy_detector_before_interpolation.bad_by_correlation,
        'bad_by_SNR' : reference.noisy_detector_before_interpolation.bad_by_SNR, 
        'bad_by_dropout' : reference.noisy_detector_before_interpolation.bad_by_dropout,
        'bad_by_ransac' : reference.noisy_detector_before_interpolation.bad_by_ransac
        }

        self.noisy_detector_after_interpolation = \
        { \
        'bad_by_nan' : reference.noisy_detector_after_interpolation.bad_by_nan,
        'bad_by_flat' : reference.noisy_detector_after_interpolation.bad_by_flat,
        'bad_by_deviation' : reference.noisy_detector_after_interpolation.bad_by_deviation,
        'bad_by_hf_noise' : reference.noisy_detector_after_interpolation.bad_by_hf_noise,
        'bad_by_correlation' : reference.noisy_detector_after_interpolation.bad_by_correlation,
        'bad_by_SNR' : reference.noisy_detector_after_interpolation.bad_by_SNR, 
        'bad_by_dropout' : reference.noisy_detector_after_interpolation.bad_by_dropout,
        'bad_by_ransac' : reference.noisy_detector_after_interpolation.bad_by_ransac
        }

```

I do know that this is somewhat cumbersome to the eyes but I believe this helps greatly to examine the pyprep with more detail without having to do debug to get inside the functions.

Looking forward to your feedback.

sappelhoff, 2020-04-07:
> We should be able to see the result of each criteria in the prepPipeline object

I absolutely agree :+1: PREP should be transparent at every step.

Your proposed implementation seems like a quick fix, however I think that we can get the code a bit cleaner by doing a more full refactoring, like @adam2392 said he might have some time for in one or two weeks.

yjmantilla, 2020-12-01:
@sappelhoff Although this is a quick fix, could we implement this to make the pyprep validation against matlab easier.

sappelhoff, 2020-12-02:
agreed @yjmantilla if it helps debugging/inspecting pyprep we should implement that now ... and refactor later, once the critical issues are solved. PR is welcome!

PR #45 (2020-12-09): added noisy_channels_before and _after interpolation attributes
Thanks for contributing.
If this is your first time, please make sure to read the [contributing guideline][contrib-guideline].

# PR Description

This PR adds two attributes to PrepPipeline:

- noisy_channels_before_interpolation
- noisy_channels_after_interpolation

Which are dictionaries with the detailed results of bad channels identified by each criteria, that is:

```python
noisy_channels_x_x = {
            "bad_by_nan": noisy_detector.bad_by_nan,
            "bad_by_flat": noisy_detector.bad_by_flat,
            "bad_by_dev [... cut]
closed 2020-12-09 by sappelhoff