sappelhoff/pyprep #21: Regarding the validation example

yjmantilla, 2020-04-07:
Regarding the validation example. I saw the plot containing the differences across channels and times. It is helpful to see the overall differences in all the channels. 

Now I think it would be helpful to see the signals directly to get a more intuitive grasp of how much difference is there. Why this?

Back when I tried to do my own implementation of prep in python I developed some functions that would compare two signals. Following the example they made I see they did a kind of relative difference against the max value of the EEG. When I did mine I divided by the value of the matlab EEG in that point, this caused that when the signal of matlab was near zero I would get big errors, to solve this problems I searched for various methods and found this thread: [https://stats.stackexchange.com/questions/86708/how-to-calculate-relative-error-when-the-true-value-is-zero](url) . So I added the relative percent difference and some others.

Point is, I was still not getting a real sense of what the difference was so I made a function that would get the difference vector and automatically plot the signals in the neighborhood of the worst difference. This would help me assess better whether the differences were significant or not.

The original plot implemented in the example that I get when running it:

[image]

What follows are the plots I get using my error plotting functions (the data itself is the same) with the relative percentage difference (RPD) method from the link I mentioned. I provide for each stage the error matrix image (channels x time) and an example of the signals plotted near the greatest difference points.

 ### NOTICE: RPD value is always between 2 and -2 thus should be interpreted in that range

## Raw Signal:

[image]

Here the signal is exactly alike so no signals plot here. (Notice the 1e-16 scaling).

## New Signal: (Detrend)

[image]

[image]

## Clean Signal (Line Noise Filter)

[image]

[image]

## Ref Signal (Referencing): (maybe related to #19 ?--> Apparently NO, This pipeline was run with a code that already fixed that one)

[image]

[image]
[image]
[image]

## Interpolated Final Signal : 

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

In this last one the difference is quite high across the channels x time points. Some high differences are expected because prep and pyprep didn't interpolate exactly the same channels, so those channels not agreed by both will have a high difference. 

# Disclaimer

Depending on how one measures the difference the error matrix image (channels x time) and the choice of the color map can look quite different and one may freak out or not depending on that. 

For example in the final stage (interpolation) we have the following plots that doesn't look at scary as the relative percentage difference one, these were obtained by other methods.

## Unsigned RPD (0 to 2)

[image]

## Unsigned Relative to the mean value (this is %):

[image]

**Notice axis goes up to 700%**

## Unsigned Relative to the max value (this is %)

[image]
**Notice axis goes up to 80%**

## Squared Error divided by max (Similar to the method used in the example)
[image]

Nevertheless all refer to the same signals so given that I think is useful to take a look at the signals themselves to evaluate the significance of the differences.

jasmainak, 2020-04-07:
I didn't read carefully and I'm not sure whether @sappelhoff did an independent implementation of RANSAC or copied parts from autoreject. But I should point out that the Matlab code is buggy. When I re-implemented it for comparison with autoreject, I dug into what was driving the differences. I fixed the random seeds etc. and eventually found that the way they compute the median was wrong. Instead of using a standard function, they had something that worked only for odd/even number of samples. Not sure it's relevant, so ignore me if it's not :)

yjmantilla, 2020-04-07:
Reading your ransac autoreject code it seems @sappelhoff  did his own implementation. But this is relevant because here we are comparing as if the matlab results were the correct ones so what you point out maybe be yet another source from the differences.

Reading this made me remember that when I examined the ransac in matlab I noticed that somehow it was really consistent across runs even if I forced the random generator to use the time. Maybe it was the particular eeg I used for that (didnt really test that issue much)... But anyhow this illustrates the fact that we should treat the comparison with matlab with a pinch of salt.

From the testing I have done it seems that pyprep implementation usually finds a bit more of bad channels and it may get different results across runs, like one channel will some times appear and some others it will not, which makes sense since ransac is random and we have not put a particular seed for it.

This also reminds me that @sappelhoff  did have a bit of different parameters for the bad channels criteria;  when @AamnaLawrence  and @Nick3151  did their implementation they put the original PREP ones but now I do wonder if @sappelhoff  had a better argument for the parameters he originally used.

Anyhow thanks for stopping by.

sappelhoff, 2020-04-07:
> I didn't read carefully and I'm not sure whether @sappelhoff did an independent implementation of RANSAC or copied parts from autoreject

I did my own implementation inspired by @jasmainak's code and the Matlab code and the respective papers ...

>  But I should point out that the Matlab code is buggy. 

Did you ever open an issue over at their repo @jasmainak ?

> But anyhow this illustrates the fact that we should treat the comparison with matlab with a pinch of salt.

I agree

> like one channel will some times appear and some others it will not, which makes sense since ransac is random and we have not put a particular seed for it.

I find that a bit troubling ... the randomness in ransac should not have *such* a big effect as changing a channel from bad to good or vice versa. UNLESS of course: ransac has few iterations, OR that particular channel is really borderline between good and bad

> if @sappelhoff had a better argument for the parameters he originally used.

I only ever implemented the "find_noisy_channels" functionality (described in [this example](https://pyprep.readthedocs.io/en/latest/auto_examples/use_noisy_module.html#sphx-glr-auto-examples-use-noisy-module-py) and [here in the API](https://pyprep.readthedocs.io/en/latest/api.html) .. it's all via the "noisydata" class) ... and for that I tried to stay as close to the paper as possible at the time (given time constraints)

---

Overall, what I take from this issue:

1. improve the [full PREP](https://pyprep.readthedocs.io/en/latest/auto_examples/run_full_prep.html#sphx-glr-auto-examples-run-full-prep-py) example, see also #15  ... next to splitting (a) showing functionality and (b) validation, specific care should be directed towards (b), with lots of explanatory text and better visualizations.

1. we should test our RANSAC against autoreject RANSAC --> could be done in the form of a sphinx-gallery example, similar to the two we already have.

jasmainak, 2020-04-07:
I think [this](https://github.com/VisLab/EEG-Clean-Tools/blob/48c2e670cf47ba998edae45a9b49c74595bddb3a/PrepPipeline/utilities/findNoisyChannels.m#L352-L353) is the lines in question if you want to follow up. I don't think I ever ended up opening an issue.

> like one channel will some times appear and some others it will not, which makes sense since ransac is random and we have not put a particular seed for it.

I would suggest actually fixing the seed to be the same in Matlab and Python so you can compare exactly.

yjmantilla, 2020-04-07:
> I would suggest actually fixing the seed to be the same in Matlab and Python so you can compare exactly.

Yes, I could implement this in #17 too right away.

sappelhoff, 2021-06-23:
@yjmantilla I think this issue can be closed, because it has been superseded by @a-hurst's https://github.com/sappelhoff/pyprep/issues/58 --> what do you think?

yjmantilla, 2021-06-23:
I agree , closing it

closed 2021-06-23 by yjmantilla