sappelhoff/pyprep #27: Pyprep drops eog channels from mne.info

SebastianSpeer, 2020-05-28:
Dear pyprep-team,

I've used pyprep for preprocessing my EEG data and now want to do additional preprocessing steps including ICA. For ICA I want to use the EOG channel as a reference channel to select ICA components. Unfortunately, during preprocessing the eog channels were dropped from the info structure and it does not seem to be possible to add the back. Is this an issue with the pyprep pipeline or is there a simple workaround?

Before pyprep:

`<Info | 8 non-empty values
 bads: []
 ch_names: Fp1, AF7, AF3, F1, F3, F5, F7, FT7, FC5, FC3, FC1, C1, C3, C5, ...
 chs: 64 EEG, 4 EOG
 custom_ref_applied: False
 dig: 67 items (3 Cardinal, 64 EEG)
 highpass: 0.0 Hz
 lowpass: 104.0 Hz
 meas_date: 2018-04-06 09:34:26 UTC
 nchan: 68
 projs: []
 sfreq: 512.0 Hz
>`

After pyprep:

`<Info | 8 non-empty values
 bads: []
 ch_names: Fp1, AF7, AF3, F1, F3, F5, F7, FT7, FC5, FC3, FC1, C1, C3, C5, ...
 chs: 64 EEG
 custom_ref_applied: False
 dig: 67 items (3 Cardinal, 64 EEG)
 highpass: 0.0 Hz
 lowpass: 50.0 Hz
 meas_date: 2018-04-06 09:34:26 UTC
 nchan: 64
 projs: []
 sfreq: 100.0 Hz
>`

sappelhoff, 2020-05-28:
Hi @SebastianSpeer can you post that part of the code where you call pyprep?

Markdown tip:

If you want to format code over multiple lines, you need to use three backticks at the start, and at the end:

like this:

````
```
#code goes here
```
````

To also add coloring to your code snippet, you can add the language to be used for coloring to the top line, e.g.

````Python
```Python
def my_python_func(a,b):
    return a**b
```
````

for code on a single line, you can use single backticks, like you did:  `hello this is code`

yjmantilla, 2020-05-28:
I think you should be able to copy the eog channels to another object (in example copying the raw object and then dropping all but eog); then add them back to the raw object thats preprocessed with raw.add_channels().

Now there may be problems with the times (ie the array length)? Im not sure, I think there should not be problems. But thats one thing to try, never done it though.

The dropping inside pyprep is indeed done in the constructor of the PrepPipeline class. Incidentally maybe inside prep we should do that to not lose the eog channels when we return the preprocessed object.

SebastianSpeer, 2020-05-28:
> I think you should be able to copy the eog channels to another object (in example copying the raw object and then dropping all but eog); then add them back to the raw object thats preprocessed with [...]

 @yjmantilla  Thanks for your suggestion. I've tried this but indeed the dimensions do not seem to match:

```
eeg_index = mne.pick_types(raw.info, eeg=True, eog=False, meg=False)
ch_names = raw.info["ch_names"]
ch_names_eeg = list(np.asarray(ch_names)[eeg_index])
raw_without_eeg=raw.copy().drop_channels(ch_names_eeg)
prep.raw.add_channels([without_raw])
```
[image]

yjmantilla, 2020-06-01:
@SebastianSpeer did you make another comment here? Because in my email notifications it seems like you commented something else but here is nowhere to be found.

SebastianSpeer, 2020-06-01:
@yjmantilla I initially commented something, but then I figured it out myself. Your suggestion worked fine. Thanks!

closed 2020-06-01 by SebastianSpeer