sappelhoff/pyprep #33: EOG get dropped

christian-oreilly, 2020-08-04:
I suppose most people would want to use PyPrep as an early preprocessing pipeline to get ride of line noise and get a robust average reference (at least, that is my understanding from the documentation). To do that, it seems that PyPrep should make available a raw object that has been cleaned and re-referenced. My understanding is that it does, as the .raw attribute of the PrepPipeline class. However, in this Raw object, some channels (like EOG channels) are dropped by PyPrep. I don't think that they should be, since they are likely to be necessary for the following artifact rejection steps that will be run after PyPrep.

sappelhoff, 2020-08-04:
Hey @christian-oreilly I didn't write that part of the code, but it's most likely for convenience, because only EEG channels are "corrected" in the PREP flow.

But there is certainly a way to keep the EOG channels part of raw and simply ignore them throughout the flow. A PR to achieve that would be welcome!

yjmantilla, 2020-08-04:
I was thinking of this just yesterday. Probably at the end we could just add the missing channels. Alternatively the reference and rereference channel parameters that get into pyprep should be able to do that but im not sure of the current behavior down to the lowest level.

christian-oreilly, 2020-08-04:
PR #34 would do it. I made it before seeing @yjmantilla comments so I will let you check if it does it how you want it to be done but I think it is well-aligned with the direction you were proposing,i.e., make a copy of the non-eeg channels and merge them back at the end... but I did not do it in the .raw object directly in case it can have side effect. I made it available through a property that add on the fly the missing channels to the .raw object and return a copy.

yjmantilla, 2020-08-04:
I think the way you did it is nice. Hopefully after the formatting changes it should complete the tests successfully

PR #34 (2020-09-01): Adding a full_raw property to PrepPipeline
Fix #33.
closed 2020-09-01 by sappelhoff