sappelhoff/pyprep #35: memory consuming

guishuyunye-lyw, 2020-08-12:
MemoryError: Unable to allocate 9.75 GiB for an array with shape (6545, 100001) and data type complex128

It just cannot work on a big dataset, which is just 500mb.

yjmantilla, 2020-08-12:
My intuition is that this comes from the multitaper (The ransac would throw another error ). Please refer to #18 . Specifically you need a patch for the multitaper described in there or alternatively apply another filtering method for the line noise while passing an empty list to the line_freqs parameter.

sappelhoff, 2020-08-12:
We basically need MNE-Python `master` with the window-wise spectrum filter and then @christian-oreilly's patch to include that in pyprep.

guishuyunye-lyw, 2020-08-13:
How to make it happen? Could you offer some specific alternative codes or example for us to imitate temporarily.

guishuyunye-lyw, 2020-08-13:
> My intuition is that this comes from the multitaper (The ransac would throw another error ). Please refer to #18 . Specifically you need a patch for the multitaper described in there or [...]

I just made the line_freqs=[0], but it still broken off. Could you give me an example how to achieve what you described. THX so much!

yjmantilla, 2020-08-13:
@guishuyunye-lyw rather than making line_freqs = [0], make it empty (line_freqs = []) . That will skip the line noise filtering.

Now if you do have line_noise you have two options. Either make a notch filter with mne for the line noise  (https://mne.tools/stable/auto_tutorials/preprocessing/plot_30_filtering_resampling.html?highlight=notch) outside of the pyprep (preferably before) or use the patch of #18 . 

I havent implemented the patch myself but I think it is already merged in MNE so you would need to install mne from the latest code ( https://github.com/mne-tools/mne-python for instructions). I think that would be enough along with passing the line_freqs you want. We are still unsure how much it actually attenuates the line noise but at least it wont explode in ram consumption.

There is another patch made by @christian-oreilly that (as far as i know) helps in case you want to keep non eeg channels in your data (ie ocular channels).

yjmantilla, 2020-08-24:
@guishuyunye-lyw  were you able to do it? can we close this?

sappelhoff, 2020-09-28:
@yjmantilla now that MNE-Python 0.21 is out, do you have time to make a PR against pyprep to set the spectrum filter to default to work in windowed mode (with 10s windows by default)? And perhaps test whether that'd a good idea?

We could also add a new parameter `filter_kwargs` which would be a dictionary that, if supplied, gets passed on the the spectrum filter call. that way, users could tweak the spectrum filter to their own liking more completely ... and if they don't do that, the default values should be reasonable.

WDYT?

yjmantilla, 2020-09-29:
@sappelhoff next week's tuesday I will get rid of an university exam and by then I will probably have the time to make the PR.

sappelhoff, 2020-09-30:
okay cool, note also this (from the changelog of mne 0.21):

> The default window size set by filter_length when method='spectrum_fit' in mne.io.Raw.notch_filter() will change from None (use whole file) to '10s' in 0.22, by Eric Larson

https://mne.tools/stable/whats_new.html#whats-new

PR #40 (2020-10-10): Adopt windowed approach to spectrum fit notch filtering as the new default
# PR Description

- Adds the defaults 10s filter length
- Adds filter_kwargs parameter

closes #35 #18 

I dont know if this is what you have in mind...

I'm also a bit busy again ( new exams, deadlines approaching etc) but I managed advance this a bit.

What do you mean by (in #35) 

> And perhaps test whether that'd a good idea?

Originally clean line uses like 2-4 second windows with 50% overlap.
closed 2020-10-10 by sappelhoff