sappelhoff/pyprep #52: "Undefined label" warning using intersphinx and numpydoc

sappelhoff, 2021-01-21:
I just today realized that to make intersphinx and numpydoc work together, on needs to set  `numpydoc_xref_param_type = True
` in `conf.py`.

I did that, and it works in many other projects ... but somehow not in `pyprep`. when building the docs, I get these warnings (see below)

@larsoner, given your expertise with both numpydoc and sphinx - do you know what's going on? why is intersphinx not detecting the "bool" type?

link to docs where links are broken: https://pyprep.readthedocs.io/en/latest/generated/find_noisy_channels.NoisyChannels.html#find_noisy_channels.NoisyChannels

```
/find_noisy_channels.py:docstring of find_noisy_channels.NoisyChannels.__init__:12: WARNING: undefined label: python:bltin-boolean-values
/find_noisy_channels.py:docstring of find_noisy_channels.NoisyChannels.find_all_bads:22: WARNING: undefined label: python:bltin-boolean-values
/find_noisy_channels.py:docstring of find_noisy_channels.NoisyChannels.get_bads:23: WARNING: undefined label: python:bltin-boolean-values

/find_noisy_channels.py:docstring of find_noisy_channels.NoisyChannels.get_ransac_pred:14: WARNING: term not in glossary: numpy:array_like
/find_noisy_channels.py:docstring of find_noisy_channels.NoisyChannels.ransac_correlations:17: WARNING: term not in glossary: numpy:array_like
/find_noisy_channels.py:docstring of find_noisy_channels.NoisyChannels.run_ransac:16: WARNING: term not in glossary: numpy:array_like
/prep_pipeline.py:docstring of prep_pipeline.PrepPipeline:32: WARNING: undefined label: python:bltin-boolean-values
```

numpydoc config: https://github.com/sappelhoff/pyprep/blob/e392f5ace47db971b0fd1a45024b600e6b95f4de/docs/conf.py#L49-L52

interspinx config: https://github.com/sappelhoff/pyprep/blob/e392f5ace47db971b0fd1a45024b600e6b95f4de/docs/conf.py#L93-L100

example of what's apparently broken: https://github.com/sappelhoff/pyprep/blob/e392f5ace47db971b0fd1a45024b600e6b95f4de/pyprep/find_noisy_channels.py#L28-L44

see the corresponding error:

> /find_noisy_channels.py:docstring of find_noisy_channels.NoisyChannels.__init__:12: WARNING: undefined label: python:bltin-boolean-values

larsoner, 2021-01-21:
Do you have this problem on both the latest `numpydoc` release and master?

sappelhoff, 2021-01-21:
Yes, on most recent stable (`pip install -U numpydoc`) *and* master (`pip install git+https://github.com/numpy/numpydoc/#egg=numpydoc` --> `1.2.dev0`)

larsoner, 2021-01-21:
Nothing immediately jumps out -- I would take your `docs` dir and whittle it down to the tiniest `conf.py` that still causes the problem. Or you can do the reverse and start from [tinybuild](https://github.com/numpy/numpydoc/tree/master/numpydoc/tests/tinybuild) in `numpydoc` (which is already very quick to build and check with `make` in the directory) to see what you need to add/change to get it to break. Somewhere between these two, things will break.

sappelhoff, 2021-01-21:
alright, thanks for having a look! I'll try to pin it down.

sappelhoff, 2021-01-21:
After more fiddling I found the issue --> I simply forgot to explicitly import intersphinx in conf.py --> I didn't notice this because *some* links did work (probably for other reasons). Now it's working.

closed 2021-01-21 by sappelhoff

akaszynski, 2023-05-23:
Encountered this issue recently and it was because my intersphinx was configured to use the "dev" documentation for python rather than the stable one. Check if your `conf.py` has:

```py
intersphinx_mapping = {
    "python": ("https://docs.python.org/dev", None),
}
```

If so, change to:

```py
intersphinx_mapping = {
    "python": ("https://docs.python.org/3", None),
}
```

RobPasMue, 2023-10-05:
Error is showing up again now that Python 3.12 is out and version ``3`` is the latest stable one (i.e. 3.12)

RobPasMue, 2023-10-05:
Temporary fix until intersphinx/numpydoc updates properly

https://github.com/ansys/review-bot/pull/108

Fix your URL to Python 3.11 docs instead of the latest stable Python 3 docs (3.12)