sappelhoff/pyprep #55: removeTrend produces different values in PyPREP than MATLAB PREP

a-hurst, 2021-04-02:
Splitting this off from #49 to keep things organized. Basically, the removeTrend function (which essentially just runs a highpass filter to remove frequencies below 1 Hz) produces different values in PyPREP's implementation (by default, a wrapper around `mne.filter.filter_data`) than in MATLAB PREP's implementation (by default, a wrapper around EEGLAB's `pop_eegfiltnew`).

For comparison I loaded the same .set file into both EEGLAB and MNE, applied `removeTrend` to both, then re-saved the .set from EEGLAB so I could load it into MNE and compare. The PSD plots for both look identical to me, but the numeric values themselves via `raw._data` look quite different:

<details>
  <summary>Comparisons of MATLAB and PyPREP values (click to unhide)</summary>

First sample of first 10 channels in MATLAB PREP before trend removal:
```
signal.data(:, 1)

ans =

   1.0e+04 *

   -0.3335
    1.0037
   -0.3581
   -0.1795
    0.9913
   -0.7414
   -0.7198
   -0.5665
    0.5557
   -1.2237
```

 First sample of first 10 channels in MATLAB PREP after trend removal:
```
signal2.data(:, 1)

ans =

   -3.7068
   -4.8012
   -0.8008
   -2.5809
   -1.5205
    0.1093
    1.9928
   -4.6906
  -10.2859
    1.6814
```

First sample of first 10 channels in PyPREP before trend removal:
```
raw_copy._data[:10, 0]
array([-0.00333541,  0.01003745, -0.00358109, -0.00179516,  0.00991251,
       -0.00741443, -0.00719762, -0.00566537,  0.00555661, -0.01223686])
```

First sample of first 10 channels in PyPREP after trend removal:
```
raw_filtered._data[:10, 0]
array([-4.11658012e-19, -1.79973325e-18,  6.98802181e-19,  2.89261751e-19,
        2.50128829e-18, -9.28348110e-19,  7.57247455e-19,  3.55753838e-19,
        3.82858892e-19, -7.01343280e-19])
```
</details>

Since the plots look the same the differences might not practically matter all that much, but if we want to be able to do formal (or informal) 1:1 comparisons of PyPREP results to MatPREP, having identical results from this function would be helpful, especially since this always happens before any of the NoisyChannels operations.

yjmantilla, 2021-04-03:
Yeap, if the plot looks the same I think this is not much of an issue. See #21 , there I took a long rambling about the numerical comparisons.

I actually did some local translations of pop_eegfiltnew, Im not sure it is really worthwhile but i can share them to you if you would like that.

sappelhoff, 2021-04-04:
Interesting, IMHO if there is "not much of a difference / impact", then I think we should try to follow the matprep "style". That is, adjust MNE's filter to be in line with EEGLAB's filter ...

Unless it's too much of an effort ... or too complicated / non-robust - then I think we should stay with what we have.

a-hurst, 2021-04-08:
@sappelhoff @yjmantilla Just did a sanity check now that the quantile PR is written, and it looks like the the median channel correlations from `find_bads_by_correlation` still differ slightly depending on whether MATLAB's removeTrend or PyPREP's removeTrend is used:

<details>
  <summary>Median maximum correlations for each channel (MATLAB removeTrend):</summary>
(Note that these match MATLAB's internal values for this step exactly)
```
array([0.81084072, 0.73898046, 0.79686453, 0.87395045, 0.82583832,
       0.74521084, 0.71704812, 0.65049225, 0.65576578, 0.76237385,
       0.87346735, 0.84432825, 0.77118926, 0.52414329, 0.54862151,
       0.59430692, 0.71217292, 0.78406237, 0.81565802, 0.84976483,
       0.82877567, 0.76908228, 0.80602873, 0.7651207 , 0.84968468,
       0.85655523, 0.86861435, 0.8555764 , 0.91003147, 0.87882915,
       0.8902305 , 0.84410764, 0.80544457, 0.77551453, 0.81546514,
       0.80657217, 0.73123314, 0.8587528 , 0.8656426 , 0.86103369,
       0.82397267, 0.83116133, 0.83602286, 0.82109135, 0.82314376,
       0.8735833 , 0.89586321, 0.85625826, 0.85483782, 0.76672159,
       0.75090935, 0.76320457, 0.78835426, 0.77933322, 0.78980044,
       0.82391842, 0.85506462, 0.83078717, 0.813171  , 0.85525523,
       0.79358095, 0.88653423, 0.86941076, 0.91209381])
```
</details>

<details>
  <summary>Median maximum correlations for each channel (PyPREP's removeTrend):</summary>
```
array([0.8108905 , 0.73900619, 0.79691305, 0.8740128 , 0.82567605,
       0.74517421, 0.71705557, 0.65046776, 0.65599907, 0.76243479,
       0.87341694, 0.84433745, 0.77109291, 0.52409959, 0.54863092,
       0.59432957, 0.7121899 , 0.78405464, 0.81559417, 0.84968288,
       0.82877201, 0.76913351, 0.80596963, 0.76527194, 0.84993409,
       0.85656695, 0.86855304, 0.85568207, 0.90999021, 0.87888549,
       0.89014018, 0.84408482, 0.80550482, 0.77546254, 0.81543663,
       0.80663478, 0.73109613, 0.8587718 , 0.86557297, 0.86108103,
       0.82389078, 0.83111796, 0.83613023, 0.82110613, 0.82314311,
       0.87359905, 0.89586031, 0.8562218 , 0.85493777, 0.76665405,
       0.75068414, 0.76317537, 0.78836499, 0.77937937, 0.7898612 ,
       0.82388245, 0.85499313, 0.8307775 , 0.8132598 , 0.85530947,
       0.79363585, 0.8865627 , 0.86948814, 0.91199044])
```
</details>

As you can see, the values are only off by ~0.001 but it still causes difficulties for MATLAB v. PyPREP comparisons, especially if we want to eventually do them directly in CI.

sappelhoff, 2021-04-08:
so are you suggesting that there is - apart from the quantile "bug" - another bug? Or is this rather numerical differences between some low level programming language functions?

a-hurst, 2021-04-08:
I'm suggesting that there's some minor difference in the settings or implementation of MNE's FIR filter vs EEGLAB's which leads to slightly different results. The main thing I've seen to suggest this is that the verbose input for both suggests that MNE's filter has a slightly different filter length than EEGLAB's.

a-hurst, 2021-04-12:
Just as a quick note on this before I forget: the 50 Hz lowpass filter that happens as part of the HF noise detector in NoisyChannels produces identical results to the one in MATLAB PREP. The main difference I can see from the detrend (other than the filter type) is that the 50 Hz lowpass is done in PyPREP with a custom `pyprep.utils.filter_design` function.

@sappelhoff I'm having some trouble making sense of `filter_design`'s docstring, but based on the code do you think it could be used to do the 1 Hz highpass detrend? If so, do you know what syntax I'd need to use for that? If it matches EEGLAB with the lowpass, maybe it'll match it with the highpass as well.

yjmantilla, 2021-04-12:
Theoretically one could do spectral inversion I think. 

```
n = 101
a = signal.firwin(n, cutoff = 0.3, window = "hanning")
#Spectral inversion
a = -a
a[n/2] = a[n/2] + 1
mfreqz(a)
show()
```

From: https://mpastell.com/2010/01/18/fir-with-scipy/

(I think the filter "kernel" are the coefficientes , but Im not sure)

> As you can see, the values are only off by ~0.001 but it still causes difficulties for MATLAB v. PyPREP comparisons, especially if we want to eventually do them directly in CI.

Using np.isclose with a 0.001 tolerance wouldn't help here?

sappelhoff, 2021-04-12:
> The main difference I can see from the detrend (other than the filter type) is that the 50 Hz lowpass is done in PyPREP with a custom pyprep.utils.filter_design function.

yes, I saw that as well - but I have never used this myself. That part of the code was written by @AamnaLawrence and @Nick3151  as far as I remember.

Nick3151, 2021-04-12:
@a-hurst Hi, sorry for not making the docstring clear. Actually the filter_design function was not written by me... but as far as I know the function simply designs an FIR filter by giving the frequency responses(amplitude) at a few given frequencies and returns the kernel(which shoule be the coefficients in time domain)

This is a funcion which we think could skip the intrinsic python implementation of a filter and may generate outputs consistent with Matlab's version. 

Hope it helps, good luck!

a-hurst, 2021-04-26:
@sappelhoff @yjmantilla Figured this out! Basically, EEGLAB's filtering math calculates the number of filter samples slightly differently than MNE: giving a length of 847 instead of MNE's 845. Unfortunately it's not as simple as passing the MATLAB number to `mne.filter.filter_data`'s `filter_length` argument, since for whatever reason (seems deliberate?) that just creates a filter of length 845 and pads both ends with zeros to get 847.

Anyway, if you calculate the filter order and cutoff frequencies using MATLAB's logic and pass those to SciPy's `firwin` (what MNE uses internally anyway), it generates an identical filter to MatPREP that produces identical results when applied to the EEG data! Putting together a PR now.

With this fix in place, comparing the full re-reference results between MatPREP and PyPREP will be much easier.

PR #71 (2021-04-29): Make removeTrend output identical to MATLAB PREP

# PR Description

Closes #55.

Basically, MNE and EEGLAB calculate FIR filter parameters slightly differently (specifically, the length of the filter). To bypass this, I've added a new function that calculates the parameter values the same way that EEGLAB does and uses SciPy's `signal.firwin` to turn them into a MatPREP-identical filter. Then, I use the same internal method MNE's `filter.filter_data` does to apply the filter to the signal, producing identical output to MatPREP's `removeTrend`.

# Merge Checklist

- [x] the PR has been reviewed  [... cut]
closed 2021-04-29 by sappelhoff