sappelhoff/pyprep #56: Bad channels by correlation differ between PyPREP and MATLAB PREP

a-hurst, 2021-04-02:
In my recent comparison efforts, I noticed that the numbers of initial bad channels by correlation sometimes differed slightly between PyPREP and MATLAB PREP. Since this code doesn't involve any random number generation, it should always be consistent between runs and also between implementations.

Anyway, I did some digging with breakpoints and think I've figured it out. It has to do with [these lines here](https://github.com/sappelhoff/pyprep/blob/8476ea5c1f80c60e564ce8e4c2fca34464515b40/pyprep/find_noisy_channels.py#L320-L332):

```python
        for k in range(0, w_correlation):
            eeg_portion = np.transpose(np.squeeze(EEG_new_win[:, :, k]))
            data_portion = np.transpose(np.squeeze(data_win[:, :, k]))
            window_correlation = np.corrcoef(np.transpose(eeg_portion))
            abs_corr = np.abs(
                np.subtract(window_correlation, np.diag(np.diag(window_correlation)))
            )
            channel_correlation[k, :] = np.quantile(abs_corr, 0.98, axis=0)
            noiselevels[k, :] = np.divide(
                robust.mad(np.subtract(data_portion, eeg_portion), c=1),
                robust.mad(eeg_portion, c=1),
            )
            channel_deviations[k, :] = 0.7413 * iqr(data_portion, axis=0)
```
Using the same input data, we see that `eeg_portion`, `data_portion`, `window_correlation`, `abs_corr`, and `noiselevels` are all identical between PyPREP and MatPREP, but `channel_correlation` and `channel_deviations` both differ.

The reason for this is that apparently [MATLAB's quantile functions work slightly differently than Numpy's and SciPy's](https://stackoverflow.com/a/58424860): instead of having quantiles be restricted to the range [0, 1], they get restricted to the range [0.5 / n, (n - 0.5) / n], with n being the number of samples passed to the quantile function. This is apparently equivalent to treating the provided values as a sample of an unknown population rather than a full population themselves. After hacking together some new `quantile` and `iqr` functions that account for this, the `channel_correlation` and `channel_deviations` data both match the MATLAB values!

Anyway, I'll put together a PR to address this, but I figure there should be a discussion of which quantile logic actually makes the most sense for this use-case. If Numpy's default logic is actually more reasonable/justifiable here, we can make the new quantile functions optional with a `match_matlab` flag  or parameter of some kind indicating strict adherence to the original PREP logic. If MATLAB's logic makes more sense, I'm assuming we should just fully replace `np.quantile` and `iqr` with their modified equivalents?

yjmantilla, 2021-04-03:
Wow, you are really digging inside. Looking forward to this.

a-hurst, 2021-04-03:
Heh, well I figure it's probably a sunk-cost thing at this point: I've dedicated enough hours of my life to trying to get to the bottom of this that I feel like I need to see this through!

Now that I have a rough understanding of how MATLAB works and how to run PREP with it (thanks again for your example script btw) I can finally do all the low level step-by-step comparisons I've always wanted to, so hopefully reaching result-parity with MATLAB PREP will happen a lot quicker now.

sappelhoff, 2021-04-04:
awesome! :tada: PR very welcome! 

> Anyway, I'll put together a PR to address this, but I figure there should be a discussion of which quantile logic actually makes the most sense for this use-case

yes, such a discussion would be valuable. For now I suggest in your PR you adjust pyprep to be in line with matprep, and we document this interesting "quantile" detail in the changelog  and/or an inline code comment to return to it sometime in the future.

PR #57 (2021-04-10): Add MATLAB-style implementations of the quantile and IQR functions
# PR Description

Closes #56. As mentioned in that issue, MATLAB's quantile and IQR functions calculate quantiles slightly differently than Numpy and SciPy, leading to higher sensitivity to differences in channel correlation than in the original MATLAB PREP. This PR adds new `mat_quantile` and `mat_iqr` functions that match MATLAB's quantile logic and make PyPREP's internal correlation matrix for `NoisyChannels.find_bad_by_correlation` match the values in MATLAB PREP's equivalent matrix exactly.

### TODO

- I wasn't sure wheth [... cut]
closed 2021-04-10 by sappelhoff