sappelhoff/pyprep #61: RANSAC channel picks differ between MATLAB PREP and PyPREP using the same seed

a-hurst, 2021-04-09:
When RANSAC does its random channel selection, the random channel choices diverge between MATLAB PREP and PyPREP, even when using the same seed. This isn't super-surprising since they're different languages with different randomization functions, but it would be handy if we could PyPREP to make the same picks as MatPREP given the same seed.

In my preliminary efforts looking into this, I've learned a few things:

- Most importantly, Numpy offers an identical implementation of the "twister" randomization algorithm to MATLAB such that they should be able to produce identical random results given the same seed
- MATLAB PREP implements its own custom `randsample` function for selecting subsets of channels without replacement. If I re-implement the same logic in Python I identical results.
- MATLAB PREP also has a `getRandomSubsets` function that calls the above `randsample` function and produces different numbers yet again. Haven't tried re-implementing it in Python yet, but hopefully it's trivial!

With some luck I'll have a PR for this shortly. It's going to break seed compatibility with earlier PyPREP versions, but I think we've already done that since last release in #43.

sappelhoff, 2021-04-09:
> With some luck I'll have a PR for this shortly. It's going to break seed compatibility with earlier PyPREP versions, but I think we've already done that since last release in #43.

yes, maintaining backwards compatibility (also with seeds) is no issue we need to burden ourselves with at this point.

a-hurst, 2021-04-09:
@sappelhoff I've got this implemented and working now (just waiting on the correlation PR to be merged before I make a new one), but in the process I've noticed something interesting that might explain some RANSAC variability:

In MATLAB PREP, the number of channels to use for RANSAC predictions is 25% of the total number of EEG channels in the data (e.g. 16 for a 64-channel dataset). In PyPREP, it's 25% of the number of *good* channels (e.g. 15 for a 64-channel dataset where 4 channels were bad by correlation). Both implementations ignore bad channels when making selections, but MATLAB PREP is going to generally be making its predictions based on more channels (and thus is likely producing better estimates). The effect of this would be compounded with each iteration of NoisyChannels during robust re-referencing, as PyPREP makes its RANSAC predictions with fewer and fewer channels each round (assuming new bad channels are detected each time).

My question is: is there a reason to prefer PyPREP's approach over MatPREP's here, or should I change it to match MatPREP's behaviour when I make my PR for this?

yjmantilla, 2021-04-10:
Dude you are on fire!

sappelhoff, 2021-04-10:
very interesting :thinking: thanks for digging this up. I am pretty sure that when I coded ransac in 2018, I tried to stay as close as possible to the publication: https://www.frontiersin.org/articles/10.3389/fninf.2015.00016/full

it's entirely possible that either:

- I misread something in the paper ... or
- the matlab PREP does not follow what it says in the paper

If paper and matlab prep both agree, then let's change pyprep.

if paper and pyprep agree, then let's first raise an issue on matlab prep to discuss.

sappelhoff, 2021-04-10:
PS: it may also be worth looking into good old github.com/autoreject/autoreject again, how Mainak did it back then.

a-hurst, 2021-04-10:
@sappelhoff Here's what the PREP publication says about it, it's sort of ambiguous:

> Before applying RANSAC, we remove channels designated as noisy by other methods. We then select a random subset of predictor channels for each channel (25% by default). If not enough channels remain to form the required subsets, the algorithm terminates. 

However, the AutoReject code seems to match MATLAB PREP here:

```python
# number of channels to interpolate from
n_samples = int(np.round(self.min_channels * n_channels))
```

What do you think? Should I add this to the next PR?

sappelhoff, 2021-04-10:
> What do you think? Should I add this to the next PR?

okay, let's do it that way then (like autoreject and matprep) :-) 

Strictly speaking, it should not make too much of a difference.

PR #62 (2021-04-10): Make PyPREP's random channel picks match MATLAB PREP's for the same random seeds

# PR Description

Closes #61. This PR does two main things:

1. Re-implements MATLAB PREP's custom `randsample` function so that we can generate identical random channel picks using the same random seed
2. Changes the RANSAC sample size from (0.25 * number of good channels) to (0.25 * number of all channels), matching MATLAB PREP and AutoReject's implementations.

I also adjusted the "too few channels" exceptions to be more informative, based on the second change. Let me know if you have any questions!

# Merge C [... cut]
closed 2021-04-10 by sappelhoff