sappelhoff/pyprep #63: PyPREP excludes more channels from RANSAC than MATLAB PREP

a-hurst, 2021-04-11:
After fixing the correlation, deviation, and channel picks code to match MATLAB PREP, I finally started trying to dig into differences in RANSAC interpolation between implementations. In the process, however, I noticed that the channel picks *still* weren't lining up between PyPREP and MATLAB PREP despite my earlier efforts.

Some deeper digging revealed that this is because PyPREP and MATLAB PREP are starting with different pools of "good" channels: PyPREP with 58 and MATLAB PREP with 60! Apparently, MATLAB PREP doesn't exclude channels bad by HF noise prior to RANSAC: only channels bad by NaN, flat, deviation, correlation, or dropout (note: NaN and flat channels already removed earlier in code):

```matlab
%% Bad so far by amplitude and correlation (take these out before doing ransac)
noisyChannels = union(noisyOut.noisyChannels.badChannelsFromDeviation, ...
    union(noisyOut.noisyChannels.badChannelsFromCorrelation, ...
          noisyOut.noisyChannels.badChannelsFromDropOuts));
```

Presumably this is because RANSAC applies a 50 Hz lowpass filter before doing correlations anyway, so HF noise wouldn't necessarily a priori grounds for exclusion. Does this make sense, or does it look like a bug in the MatPREP code? Is this a behaviour we'd like to mimic in PyPREP?

sappelhoff, 2021-04-11:
>  Apparently, MATLAB PREP doesn't exclude channels bad by HF noise prior to RANSAC: only channels bad by NaN, flat, deviation, correlation, or dropout (note: NaN and flat channels already removed earlier in code)

interesting, I would have assumed all channels previously identified as bad would be excluded.

> Presumably this is because RANSAC applies a 50 Hz lowpass filter before doing correlations anyway, so HF noise wouldn't necessarily be a priori grounds for exclusion.

that makes sense to me 🤔 is there anything on this in the paper?

a-hurst, 2021-04-11:
> is there anything on this in the paper?

I checked, and the paper gives no indication that this is happening:

>We use the RANSAC (random sample consensus) method (Fischler and Bolles, 1981) to select **a random subset of (so far) good channels** to predict the behavior of each (excluded from subset) channel in small non-overlapping time windows (5 s by default). The implementation is based on functions in BCILAB (Kothe and Makeig, 2013).
>
>The bad-by-RANSAC channels behave in a manner poorly predicted by the other channels. Before applying RANSAC, **we remove channels designated as noisy by other methods**. 

From that it sounds like HF channels would be dropped too. However, the HF noise detection procedure is only explained in the paragraph after RANSAC, so that might suggest it's not meant to be part of the pre-exclusion process. Also, I checked the git blame for those lines and they've been that way since at least 2014, with the only notable change being the addition of bad-by-dropout channels at some point ~6 years ago.

My personal instinct would be to side with the code here, since the rationale that they're being low-pass filtered at 50 Hz before RANSAC anyway (and haven't failed any of the other bad-by criteria) makes sense. The HF noise channels are still included in the final NoisyChannels output so they would still get ignored when doing the actual re-referencing. What do you think?

yjmantilla, 2021-04-12:
> My personal instinct would be to side with the code here, since the rationale that they're being low-pass filtered at 50 Hz before RANSAC anyway (and haven't failed any of the other bad-by criteria) [...]

I agree with this instinct. In any case after we manage to get really close to the original PREP we can start asking the questions of improving further the original technique. But as of now I think the goal is to match as close as possible the original PREP, right?. Unless is a really weird mistake with no explained rationale yet (like the weird median calculation jasmainak found) , I think we should side with the code.

sappelhoff, 2021-04-12:
okay, thanks both - I agree to side with the matprep code then :+1:

PR #64 (2021-04-12): Retain HF noise channels during RANSAC when not bad by any other metric

# PR Description

Closes #63. This PR modifies NoisyChannels so that channels bad by high-frequency noise (and only by HF-noise) are included in RANSAC predictions, as they are in MATLAB PREP.

The rationale for this (other than lining up more closely with MatPREP's internal logic) is that RANSAC uses a copy of the data that's been lowpass-filtered at 50Hz anyway, so high-frequency noise isn't necessarily automatic grounds for exclusion.

# Merge Checklist

- [x] the PR has been reviewed and all comments are resolved
- [x [... cut]
closed 2021-04-12 by sappelhoff