sappelhoff/pyprep #75: First-pass average referencing differs between PyPREP and Matlab PREP

a-hurst, 2021-05-04:
In MATLAB PREP, robust re-referencing starts by detecting noisy channels, calculating the median signal excluding all channels bad by NaN, bad by flat, or bad by low SNR, and then re-referencing the data using that signal. PyPREP does almost the same thing, except it leaves in channels bad by low SNR which results in numeric differences when those sorts of channels are present:

https://github.com/sappelhoff/pyprep/blob/051edd6e34360b9e6cdae39ede93a84cda895109/pyprep/reference.py#L233-L240

Based on the above code & comment, it looks like this difference is deliberate? Is there an intentional reason to include these channels when calculating a median reference? If so, I'll make it a `matlab_strict` thing and document it in the "deliberate differences" page. If not, I'll just make a PR to add those in. Looking at the MatPREP Git history, it looks like the SNR channels were a deliberate addition to the list of exclusions from the initial reference, added when the default method of calculating the initial reference signal was changed from "huber mean" (?) to median. Maybe it's worth asking Dr. Robbins about?

Anyway, if I add "bad-by-SNR" channels to the list of initial unusables in PyPREP, the initial average-referenced signal is identical to the one in MATLAB PREP.

sappelhoff, 2021-05-04:
[git blame](https://github.com/sappelhoff/pyprep/blame/051edd6e34360b9e6cdae39ede93a84cda895109/pyprep/reference.py#L233-L240) tells me that this was added by @adam2392 when he merged in the work by @AamnaLawrence and @Nick3151 back then. And apparently @yjmantilla noticed that it's odd in https://github.com/sappelhoff/pyprep/pull/32, and left the comment that you now stumbled on.

I guess it's just an oversight and fixing it now would be good. I cannot remember this being a deliberate choice - and unless somebody else chimes in, we should go ahead with a fix :)

> Maybe it's worth asking Dr. Robbins about?

I don't think so. Not unless we start running into problems after the fix.

> Anyway, if I add "bad-by-SNR" channels to the list of initial unusables in PyPREP, the initial average-referenced signal is identical to the one in MATLAB PREP.

Cool, +1

closed 2021-05-04 by sappelhoff

Nick3151, 2021-05-06:
Thank you for pointing out! It should just be an oversight. Go ahead

PR #78 (2021-05-04): Change initial re-referencing state in Reference to match Matlab PREP

# PR Description

Closes #75 and closes #76. This PR fixes two separate (but related) issues:

- Unlike PyPREP, MatPREP considers bad-by-SNR channels to be 'unuseable' for its initial average reference (in addition to bad-by-NaN and bad-by-flat channels)
- Unlike PyPREP, MatPREP only flags the above "unusable" channels for permanent exclusion from being reference channels after the first noisy channels pass. PyPREP includes bad channels detected by all methods from this pass as permanently unusable, which doesn't necessaril [... cut]