sappelhoff/pyprep #76: PyPREP starts re-referencing with more 'bad' channels than Matlab PREP

a-hurst, 2021-05-04:
In PyPREP, we do a first pass of NoisyChannels prior to initial average referencing to a) keep track of what the original channels flagged as "bad" were, and b) figure out which channels are "unusable" for median-referencing before we start the actual NoisyChannels -> interpolate bads -> re-reference loop. In PyPREP, we use the bad channels from this initial pass of NoisyChannels as the starting state for the subsequent loop (and once a channel is considered 'bad', it's permanently excluded from being a reference channel during re-referencing). See the last line here, where `self.noisy_channels` gets initialized with a copy of `self.noisy_channels_original`:

https://github.com/sappelhoff/pyprep/blob/051edd6e34360b9e6cdae39ede93a84cda895109/pyprep/reference.py#L216-L230

MatPREP, however, only adds the 'unusable' channels (i.e. those bad by NaN, flat, or SNR) to the starting state of permanently bad channels, leaving the deviation, HF noise, correlation, dropout, and RANSAC initial bads empty.

Personally I think MatPREP's behaviour makes more sense here, given that the signal hasn't necessarily been average-referenced coming into PyPREP (e.g. could be referenced to Cz) and the initial average referencing could change the "bad" status for a lot of channels (e.g. if Cz was particularly noisy). In fact, I'd go one further and say that "bad-by-flat" channels from the first NoisyChannels pass shouldn't be permanently excluded either for this same reason: if it's flat because it's a reference channel, it won't be flat anymore after the initial average referencing.

sappelhoff, 2021-05-04:
> MatPREP, however, only adds the 'unusable' channels (i.e. those bad by NaN, flat, or SNR) to the starting state of permanently bad channels, leaving the deviation, HF noise, correlation, dropout, and RANSAC initial bads empty.

I see :thinking: 

> e.g. if Cz was particularly noisy

actually, if Cz was noisy and the online reference, then no amount of post processing will get the data clean, due to the way the EEG signal is amplified during recording :disappointed:  (`signal_elec_i = (elec_i - ground) - (ref - ground)`)

But yes -> if we just pick another elec in your example, it makes total sense ... +1 to go into matlab direction here and assume that the data are not necessarily average referenced to begin with (but if they are, that shouldn't be a problem).

> In fact, I'd go one further and say that "bad-by-flat" channels from the first NoisyChannels pass shouldn't be permanently excluded either for this same reason: if it's flat because it's a reference channel, it won't be flat anymore after the initial average referencing.

Typical EEG datasets do not include the reference channel for that exact reason (it's flat) ... and when changing the reference of the data to e.g., average reference, you need to drop that (no not flat anymore) reference channel to have ["full rank"](https://en.wikipedia.org/wiki/Rank_(linear_algebra)) data again (e.g., for ICA) ... I don't think I get the math of that exactly, but you can try it yourself using https://numpy.org/doc/stable/reference/generated/numpy.linalg.matrix_rank.html:

- ... check a dataset with 64 channels, should be rank 64/64 (full)
- ... compute average reference ... should still be rank 64/64 (full)
- ... add reference channel to non-re-referenced data https://mne.tools/stable/generated/mne.add_reference_channels.html
- ... compute average reference on 65 channel data, should be rank 64/65 (not full rank --> ICA will be a problem)

so that's more or less an "aside" note that adding the reference channel after recording is (at least in my experience) not often done.

But let's imagine somebody would want to do that:

- add ref chn (flat) to data -> from 64 chns to 64 + 1 flat (65)
- pass data through pyprep
- --> want to get back 65 non-flat channels

would the ref chn **always** be bad after pyprep (in the current state of the code), or does it get a chance to "redeem" itself later and get interpolated?

Overall I'm +1 to fixing pyprep to be like matprep here, and -0.5 to go beyond what matprep does.

a-hurst, 2021-05-04:
> Typical EEG datasets do not include the reference channel for that exact reason (it's flat) ... and when changing the reference of the data to e.g., average reference, you need to drop that (no not [...]

Ahhhh okay, I think this makes sense: since the channel doesn't actually contain any new information not contained in other channels post-reference, it has nothing useful to contribute. This is another instance of me still being pretty new to the theory and conventions of EEG, so please disregard my extra suggestion here. I'll match behaviour to MATLAB PREP for my PR.

PR #78 (2021-05-04): Change initial re-referencing state in Reference to match Matlab PREP

# PR Description

Closes #75 and closes #76. This PR fixes two separate (but related) issues:

- Unlike PyPREP, MatPREP considers bad-by-SNR channels to be 'unuseable' for its initial average reference (in addition to bad-by-NaN and bad-by-flat channels)
- Unlike PyPREP, MatPREP only flags the above "unusable" channels for permanent exclusion from being reference channels after the first noisy channels pass. PyPREP includes bad channels detected by all methods from this pass as permanently unusable, which doesn't necessaril [... cut]
closed 2021-05-04 by sappelhoff