sappelhoff/pyprep #77: Interpolation of bad channels differs numerically between PyPREP and MatPREP

a-hurst, 2021-05-04:
For whatever reason, MATLAB PREP uses a different method of interpolation for channel prediction during RANSAC than it does for interpolating channels during and after re-referencing.

For RANSAC, they use their own internal interpolation code (a copy of the private interpolation code from EEGLAB's `clean_rawdata` plugin), which happens to match MNE's `_make_interpolation_matrix` exactly. For all other interpolation, they use a wrapper around EEGLAB's public interpolation code, which seems to use a different method of spherical spline interpolation (different numeric results, no reference to Perrin '89 in the docstring).

Anyway, I can't think of any a priori justification for using different interpolation logic in one place over the other and my gut is to trust MNE's code more than EEGLAB's, but regardless it's something that needs to be fixed for `Reference` to produce numerically-identical results to MATLAB PREP for #58. According to the original manuscript it seems like they comparison-tested the methods and found they gave similar correlations with actual signals?

> The PREP pipeline uses the `spherical` option of EEGLAB `eeg_interp` function for channel interpolation. This function uses Legendre polynomials up through degree 7. To test this choice, we compared this interpolation method with the `v4` option and two other spherical interpolation functions, including the one used in RANSAC. **We applied a rank sum test on the correlation of the interpolated channel with the actual channel and found no significant difference between the three spherical interpolation methods**. The `v4` option performed significantly worse. The results were not sensitive to whether a block of channels was removed in the immediate vicinity of the bad channel or whether the interpolation was done in ICA space and projected back.

Sort of confused by "there were no significant differences, but `v4` was significantly worse", though. May be worth doing our own testing.

sappelhoff, 2021-05-04:
So I think they tested 3 methods:

1. eeg_interp
2. "the one in RANSAC"
3. one more

and 1. they tested with parameters `v4` and `v7`.

Everything was about equal, but `v4` was worse than `v7` in `eeg_interp`.

The only reason I can think of why they would use different algorithms for interpolation is computation speed ... perhaps the RANSAC method is slightly faster? Does that even make sense?

Seems like for numerical equivalence another `matlab_strict`-wrapped function must be coded up :-(

a-hurst, 2021-05-04:
> The only reason I can think of why they would use different algorithms for interpolation is computation speed ... perhaps the RANSAC method is slightly faster? Does that even make sense?

My guess as to why they used the `clean_rawdata` function for RANSAC is because it just generates and returns interpolation matrices rather than doing any interpolation itself (since in RANSAC that gets done per-window using PREP's own code). The equivalent interpolation matrix function in `eeg_interp` is buried at the bottom of the file and separating it out isn't quite as straightforward, so their choice of function there makes practical sense.

As for their use of `eeg_interp` versus `clean_rawdata`'s method elsewhere, my two main ideas are:

1. They just wanted to use EEGLAB's public API instead of their own channel interpolation code for the sake of less maintenance burden
2. `eeg_interp` was generally better (though not so much that the difference was statistically significant), so they used it for all situations where it was suitable (i.e. didn't require getting the matrix alone)

a-hurst, 2021-05-04:
@sappelhoff Looking at EEGLAB's `eeg_interp` code again, it seems to differ quite a bit in terms of math from the Perrin method that MNE and PREP's RANSAC use. I tried googling around to see if I could figure out if it was based on a specific paper or anything, but no such luck. I did find a lot of papers and people on the EEGLAB mailing list incorrectly citing Perrin '89 for `eeg_interp`'s spherical method, though!

Anyway, I've emailed the original author of that file to see whether he remembers which paper his code is based off of, will report back with what I hear. In the meantime, I've more-or-less figured out the Python code to do`eeg_interp` style interpolation, though I'm going to hold off on a PR until I understand what the new method's doing well enough to documented and name variables informatively.

sappelhoff, 2021-05-04:
Haha, okay! It's great you are bringing a lot of patience to this, we surely need it.

a-hurst, 2021-05-12:
@sappelhoff After a week of no response (even opened an issue on the EEGLAB Github page) I went ahead and finished up my implementation of EEGLAB's interpolation code. From what I can tell, the main differences from MNE are that it a) uses Euclidian distances between electrodes instead of cosines when computing G for the Perrin formula (?), b) It calculates C for the Perrin formula slightly differently (uses different padding values?), and c) it average-references the signal from the good electrodes prior to interpolation and then adds the subtracted average back right after. I'll try and clean it up for a proper PR tomorrow if I have time.

Regardless, I did a bunch of reading up on EEG interpolation methods and their different formulas and performances ([this](https://www.researchgate.net/publication/286306139_EEG_potential_mapping_by_3D_interpolation_methods) was the best summary I found) and it looks like `eeg_interp`'s method is sort of a weird mix of components of barycentric interpolation and the Perrin spherical spline interpolation MNE uses. Since it doesn't seem to match up with any of the common/established methods I could find, I think this is definitely one for the `matlab_strict` flag.

That said, is there any advantage you can think of for temporarily removing the average reference of the good channels before interpolating? Would that reduce the influence of noise in the bad channels from affecting the quality of the interpolation?

sappelhoff, 2021-05-12:
Okay, what a mess - thanks for looking it up.

> Since it doesn't seem to match up with any of the common/established methods I could find, I think this is definitely one for the matlab_strict flag.

+1

> That said, is there any advantage you can think of for temporarily removing the average reference of the good channels before interpolating? Would that reduce the influence of noise in the bad channels from affecting the quality of the interpolation?

I don't get the exact process from your description, is it:

- get good elecs
- compute ave ref based on goods
- goods - ave
- interpolate bads
- (bads become goods)
- goods + ave

?

I don't know why they are doing it this way ... but if I have a relatively noisy "good" electrode (say TP9) and I want to interpolate another "bad" electrode C4 with good neighbors (all less noisy than TP9), then:

- doing it the way I described above will spread relatively more TP9 noise to C4 ...
- compared to not doing an ave ref

I don't see the advantage right now.

a-hurst, 2021-05-12:
Yep, you got it. Here's what the code looks like for reference:

```python
def _eeglab_interpolate(data, pos_from, pos_to):
    # Calculate G for distances between good electrodes, and 
    G_from = _eeglab_calc_g(pos_from, pos_from)
    G_to_from = _eeglab_calc_g(pos_from, pos_to)

    # Get average reference signal for all good channels and subtract from data
    avg_ref = np.mean(data, axis=0)
    data_tmp = data - avg_ref

    # Calculate interpolation matrix from electrode locations
    pad_ones = np.ones((1, pos_from.shape[0]))
    C_inv = linalg.pinv(np.vstack([G_from, pad_ones]))
    interp_mat = np.matmul(G_to_from, C_inv[:, :-1])

    # Interpolate bad channels and add average good reference to them
    interpolated = np.matmul(interp_mat, data_tmp) + avg_ref

    return interpolated
```

> if I have a relatively noisy "good" electrode (say TP9) and I want to interpolate another "bad" electrode C4 with good neighbors (all less noisy than TP9), then:
> 
>     * doing it the way I described above will spread relatively more TP9 noise to C4 ...
> 
>     * compared to not doing an ave ref

What if one of the bad channels is really noisy, though? See this example from my current project's dataset, processed months ago with an old version of PyPREP:

Before re-referencing:
[image]

After re-referencing:
[image]

As you can see, Fp1 is super-noisy, which ends up affecting all the other channels too due to the average referencing. If you re-reference to only the good channels when trying to interpolate Fp1, are you making a better signal because your source channels no longer have that noise? Or does it not matter, since they're adding the noisy average reference back to those channels anyway?

sappelhoff, 2021-05-14:
So if the first pic is after applying an average reference, what's the second pic after? --> is the 2nd pic process to apply pyprep after average referencing the original data?

I would just declare Fp1 as bad, and interpolate it later ... but I wouldn't calculate an average reference of good channels before interpolation and then subtract that after interpolation :man_shrugging: seems odd to me. Maybe that's for satisfying some kind of assumption they need with their custom code.

Let's copy that EEGLAB way though and wrap it in matlab_strict. And in default pyprep we can use the proper Perrin 1989 spherical spline approach that's implemented in MNE-Python. WDYT?

a-hurst, 2021-06-16:
@sappelhoff Just as an update, I've got the code this ready for a PR but it's currently soft-blocked by #73 and/or #74: without the ability to do final interpolation separately from re-referencing or having the main re-referencing code able to be`matlab_strict`, I can't add any MATLAB equivalence tests to make sure my code all lines up as intended.

To remove the roadblock, we could just wrap the seed behaviour in #74 in `matlab_strict` for now and open a new issue for testing/verifying that PyPREP's approach is indeed better (that way it can be justified in the MATLAB differences page). Otherwise, fixing #73 is probably the path of least resistance. What do you think?

sappelhoff, 2021-06-16:
> to remove the roadblock, we could just wrap the seed behaviour in #74 in matlab_strict for now and open a new issue for testing/verifying that PyPREP's approach is indeed better (that way it can be justified in the MATLAB differences page).

this sounds good to me.

PR #96 (2021-06-28): Implement EEGLAB-style interpolation in Reference for matlab_strict

# PR Description

Closes #77. This PR basically reimplements most of EEGLAB's `eeg_interp` method in Python so we can finally do MATLAB equivalence tests for the entirety of the `Reference` class (also added in this PR). This was... *not easy* to say the least, but it was necessary for the peace of mind that these MatPREP comparison CI tests allow (and also helped us identify a bunch of other small bugs/discrepancies along the way).

This is also the last numeric difference from MatPREP I'm aware of other than adaptive line n [... cut]
closed 2021-06-28 by sappelhoff