sappelhoff/pyprep #80: Dropout channel detection broken in PyPREP

a-hurst, 2021-05-05:
Working on #79, I modified `matlab_artifacts` to simulate a "dropout" channel (I set the second 25% and last 25% of the channel to 0). This channels is successfully identified by MatPREP, but in PyPREP the addition of this channel breaks NoisyChannels completely, seriously messing up the correlation windows and leaving too few usable channels to use for RANSAC.

I've been looking into it for a bit and it seems to be due to differences in how NaNs are generated during correlation between Numpy and MATLAB, but haven't figured out the exact source yet.

a-hurst, 2021-05-05:
Okay, so at least *some* of the issue is due to difference in NaN handling between Numpy's quantile functions and MATLAB's. In MATLAB, any NaNs encountered will be ignored and removed before calculation, whereas `np,quantile` just returns all NaNs if any are encountered. `np.nanquantile` does what we want it to, but then we run into problems with another Numpy/MATLAB difference. Recall that we added a custom `_mat_quantile` function to adjust the quantiles based on the size of the input sample like MATLAB does:

https://github.com/sappelhoff/pyprep/blob/ffa63467810b217635aca6f1ca0e4bfb79637b15/pyprep/utils.py#L77-L80

Because of how MATLAB handles NaNs, however, we also need to adjust the calculation of`n` here to decrease by 1 for each NaN encountered in each sample (i.e. for each completely-flat channel during the window). Maybe it's just been a long day, but I can't for the life of me figure out an elegant solution for handling this. Any ideas @sappelhoff?

sappelhoff, 2021-05-06:
I guess a first thing would be to adjust `_mat_quantile`:

```diff
+   arr = arr[~np.isnan(arr)]
    q = np.asarray(q, dtype=np.float64)
    n = len(arr)
    q_adj = ((q - 0.5) * n / (n - 1)) + 0.5
    return np.quantile(arr, np.clip(q_adj, 0, 1), axis=axis)
```

right?

I don't really see the  other problem that you are talking about :thinking: Could you explain it once more?

a-hurst, 2021-05-06:
The issue is that we're giving the function a 2-D matrix of correlation values and taking quantiles along the axis (i.e. one quantile per channel) so removing NaNs beforehand doesn't work (the change you suggested turns `arr` into a flat 1-D array). Anyway, I figured out a solution I'm reasonably happy with in the latest PR, now dropouts are getting handled smoothly!

PR #81 (2021-05-07): Fix handling of channels with dropouts (intermittent flat regions) within NoisyChannels

# PR Description

Closes #80. Because Numpy's quantile function handles NaN values differently than MATLAB, finding bad-by-correlation channels in `NoisyChannels` broke completely if there were any correlation windows where a channel had a flat signal. Also, the code that was meant to replace NaN correlation values with zeros was accidentally doing so in the wrong matrix. Both of these are fixed by this PR!

# Merge Checklist

- [x] the PR has been reviewed and all comments are resolved
- [x] all [CI][what [... cut]
closed 2021-05-07 by sappelhoff