sappelhoff/pyprep #87: fill up .zenodo.json before Release

sappelhoff, 2021-06-15:
I added a [`.zenodo.json`](https://github.com/sappelhoff/pyprep/blob/master/.zenodo.json) file to the repo in preparation for the next release, which will be archived on Zenodo.

I filled in what I already knew, but for some of you (the authors so far), the affiliation and ORCID is missing.

Please post to this thread so that we can complete the document. Please follow this example when you post here:

https://github.com/sappelhoff/pyprep/blob/7f5ed55253bbea8e064f315b1efb4c6d5974c484/.zenodo.json#L6-L8

a-hurst, 2021-06-15:
```json
"affiliation": "Dalhousie University, Halifax, Nova Scotia", 
"name": "Hurst, Austin J.", 
"orcid": "0000-0002-0219-4131"
```

Technically I'm currently affiliated with the University of Waterloo in Ontario, but I'll be officially graduated this weekend and will be starting my PhD at Dalhousie in September.

sappelhoff, 2021-06-15:
Thanks @a-hurst and congrats to your graduation and your upcoming PhD position! 🎉 

PS, if you have Twitter, let's connect there - I have a tweet where I wanted to tag you for your efforts, but couldn't: https://twitter.com/stefanappelhoff/status/1404878216469401608

yjmantilla, 2021-06-15:
Hi @sappelhoff , sorry for the late reply. Lots of things going on recently.

```
"affiliation": "Grupo Neuropsicología y Conducta, Universidad de Antioquia, Medellín, Colombia", 
"name": "Mantilla Ramos, Yorguin José",
"orcid": "0000-0003-4473-0876"
```

a-hurst, 2021-06-15:
@sappelhoff Thanks! 🙂 I actually don't have a Twitter, but I'll let you know if I ever create one: I get distracted by the internet enough as it is so I've been worried what it would to to my productivity...

Speaking of connecting though, the other day I was trying to figure out whether you could use an Raspberry Pi's GPIO for EEG triggers and stumbled on your triggers-via-Teensy preprint! Are there any supplementary materials on how to make the Teensy-based adaptor you describe, or is that still a work in progress?

sappelhoff, 2021-06-16:
> stumbled on your triggers-via-Teensy preprint!

:-) it's actually published now (open access): https://doi.org/10.3758/s13428-021-01571-z

We tried to make the supplemental material as helpful as possible, see: https://stefanappelhoff.com/usb-to-ttl/

But if you have more questions that are not answered by those docs, let me know (preferably as an issue on that page: github.com/sappelhoff/usb-to-ttl).

sappelhoff, 2021-06-16:
@christian-oreilly @AamnaLawrence @Nick3151 could you please let me know about your affiliations and ORCIDs in this format:

https://github.com/sappelhoff/pyprep/blob/7f5ed55253bbea8e064f315b1efb4c6d5974c484/.zenodo.json#L6-L8

AamnaLawrence, 2021-06-16:
Hi Stefan,

I am a masters student in biomedical engineering at John’s Hopkins university. I do not have a orcid. 

Thank you,
Aamna

christian-oreilly, 2021-06-16:
For me:
```
              {
               "affiliation": "Azrieli Centre for Autism Research, McGill University, Montreal, Canada",
                "name": "O'Reilly, Christian",
                "orcid": "0000-0002-3149-4934"
              },
```

@AamnaLawrence : FYI, you can register for one if you'd like (https://orcid.org/register). I registered back when I was PhD student if I remember well...

Nick3151, 2021-06-16:
Hi @sappelhoff --
Thank you for reaching out! 

 "affiliation": "Department of Biomedical Engineering, Johns Hopkins University", 
 "name": "Xiang, Liang", 
 "orcid": "0000-0003-3191-9565"

sappelhoff, 2021-06-16:
Thanks everybody. @AamnaLawrence if you want to register an ORCID, you can still do that and just tell me later with a reply to this thread (tagging me).

closed 2021-06-16 by sappelhoff

AamnaLawrence, 2021-06-16:
@sappelhoff/pyprep
***@***.***>
Thank you for your response.

{
               "affiliation": "Johns Hopkins University, Baltimore,
Maryland, USA",
                "name": "Lawrence, Aamna",
                "orcid": "0000-0002-0983-0997"

              }