sappelhoff/pyprep #101: Release 0.4

sappelhoff, 2021-08-02:
Hey @a-hurst @yjmantilla, I feel like we should draw up a timetable for a 0.4 release. It's not a good idea to just have the little note in the README for people to please install from `master` --> many people will not see/read it, and install from pypi/conda-forge instead, then they'll experience issues, and become disappointed and won't use pyprep.

Judging from the milestone (https://github.com/sappelhoff/pyprep/milestone/1), there are just two major things left, both pertaining to examples / documentation:

1. To overhaul (divide and conquer) the main example we have --> #15 
1. An example comparing autoreject and pyprep RANSAC --> #36 

However, more informally we discussed in #95 how we probably *should* improve the pyprep API for the 0.4 release, rather than in 0.5 --> @a-hurst has started towards this in #99 to first solve #73 (as that may simplify doing #95)

My questions are mainly:

1. @yjmantilla will you manage to finish #36, and if yes, can you estimate when?
2. @a-hurst do you have a time plan for finishing #99 and then taking a stab at #95 ?

Lastly I propose that we don't block ourselves with #15 --> if somebody has the time and energy to do it: good, else we migrate it to the 0.5 milestone.

As usual I am available for reviewing and maintaining, but probably won't have a lot of time to go into coding myself.

yjmantilla, 2021-08-03:
Hi @sappelhoff !

Sorry I have not been much active. There has been quite a lot of work on my  hands.

>@yjmantilla will you manage to finish Compare RANSAC in pyprep and autoreject #36, and if yes, can you estimate when?

Tbh I have forgot about that.  What I get reading the comments there is that the goal now is to have a documented example that compares both RANSAC's in an isolated way (that is, not as a part of PREP or Autoreject, but just as something that finds bad channels).

@a-hurst  commented that 

> Yep, this definitely seems useful! I still need to read this PR over properly so it may have this already, but it would be great to have a section in the example that compares/contrasts the usage of PyPREP vs AutoReject (i.e., types of noise detection, full file vs per-epoch) so that understanding when/where to use each is clearer to newcomers (I remember getting confused by this myself last year when first setting up my EEG pipeline).

From this what I get is that functionality whatever that PR tried to do, it is already done on another. Thus I should focus myself on a documented example.

Now, originally it was only about RANSACs, but  @a-hurst  mentions a Pyprep Vs Autoreject comparison. For me that would be two different documented examples. 

So, what is expected for 0.4 is a documented example of PyPrep RANSAC vs Autoreject RANSAC comparison (not the comparison of the whole packages). Is this correct?

Right now, I have a lot on my hands, I think I cannot promise something before October . I could advance here and there but for something guaranteed to be delivered I need my semester to finish :/  first.

sappelhoff, 2021-08-03:
> What I get reading the comments there is that the goal now is to have a documented example that compares both RANSAC's in an isolated way

Yes, I think that's what #36 could do ... just compare the pyprep and autoreject RANSAC implementations with each other.

> So, what is expected for 0.4 is a documented example of PyPrep RANSAC vs Autoreject RANSAC comparison (not the comparison of the whole packages). Is this correct?

Yep

> Right now, I have a lot on my hands, I think I cannot promise something before October . I could advance here and there but for something guaranteed to be delivered I need my semester to finish first :/

okay! :-)

a-hurst, 2021-08-03:
Hey @sappelhoff, I've been taking some time for myself and for other projects but I can definitely try to get #99 finished up in the next few days! Apart from other work, the issue is that an API-changing PR like that has a tendency to snowball as you encounter other issues and things to clean up, so I'm going to try breaking it down a bit to make things simpler/easier.