sappelhoff/pyprep #111: Passing a custom montage to the PrepPipeline(raw, montage=?)?

neerajww, 2022-02-03:
Hi,

I am working with an (open-access )EEG dataset which has 255 channels (https://zenodo.org/record/4518754#.YfukTfXTXLC). It is curry file format, and I can load the dataset in mne with its montage information.
```
 raw = mne.io.read_raw_curry( <filename> + '.dat')
```
I am confused on how to pass the "montage" information to the pyrep pipeline. I will appreciate any help with this. Excuse me if this is a naive question.

neerajww, 2022-02-03:
As a follow up, i tried with montage=None expecting the PrepPipeline will extract information from the raw object.
```
PrepPipeline(raw, prep_params, montage=None)
```
But then I get the following (different) error.
[image]

sappelhoff, 2022-02-03:
regarding your first question, see this example: https://pyprep.readthedocs.io/en/latest/auto_examples/run_full_prep.html?highlight=montage#load-data-and-prepare-it

regarding your second problem, that is a bit unexpected. Could you try saving your `raw` using [`raw.save`](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.save), and then loading it again using [`mne.io.read_raw_fif`](https://mne.tools/stable/generated/mne.io.read_raw_fif.html?highlight=read_raw_fif#mne.io.read_raw_fif)? If the error disappears then, it might be a problem in MNE-Python with the curry data format

neerajww, 2022-02-03:
Thank you for the reply. I tried raw.save() saving the file as .fif but the error persists. 
[image]

sappelhoff, 2022-02-03:
You have to save as fif AND THEN reload the data from that saved fif datafile. Only saving is not enough :-) ... but the error log looks like you did exactly that. Weird how you still get an issue.

Can you do:

```python
print(raw.get_data().shape)
```

and report what you get here?

Also can you run
```python
import mne
mne.sys_info() 
```

and report the outputs here?

Which version of pyprep are you using, and how did you install it?

neerajww, 2022-02-03:
Yes, you guessed right- I saved + reloaded - the data.
Below is the version information.
[image]

Using the preload=True solved it. Thank you!
[image]