sappelhoff/pyprep #113: Saving prep.raw give error " ValueError: Measurement infos are inconsistent for dig"

walkerped, 2022-03-26:
I am trying to data from openneuro (the first subject under https://openneuro.org/datasets/ds003690/versions/1.0.0) through pyprep with the following code.

```
import os
import mne
import numpy as np
import scipy.io as sio
import matplotlib.pyplot as plt
from pyprep.prep_pipeline import PrepPipeline

subs = ['10']
data_dir = '/mnt/f/portfolio/data/restEEG'
study_dir = os.path.join(data_dir,'ds003690')

for sub in subs:
    
    # load file
    subStr = f'sub-AB{sub}'
    sub_dir = os.path.join(study_dir,subStr,'eeg')
    eeg_file = os.path.join(sub_dir,f'sub-AB{sub}_task-passive_run-1_eeg.set')
    raw_eeg = mne.io.read_raw_eeglab(eeg_file)
    
    # Set ecg and eog
    raw_eeg = raw_eeg.set_channel_types({'EKG':'ecg','VEO':'eog','HEO':'eog' })
    raw_eeg.drop_channels(['CB1','CB2','R-Dia-X-(mm)','R-Dia-Y-(mm)'])
    
    # Extract sample rate
    sample_rate = raw_eeg.info["sfreq"]
    
    # Make a copy of the data
    prep_eeg = raw_eeg.copy()
    
    # Run prep
    prep_params = {
    "ref_chs": "eeg",
    "reref_chs": "eeg",
    "line_freqs": np.arange(60, sample_rate / 2, 60),
    }
    prep = PrepPipeline(prep_eeg, prep_params, mont1020)
    prep.fit()
```

The command appears to be successful giving the following output:

```
/tmp/ipykernel_16702/2558830226.py:9: RuntimeWarning: Unknown types found, setting as type EEG:
heog: ['HEO']
pupil: ['R-Dia-X-(mm)', 'R-Dia-Y-(mm)']
veog: ['VEO']
  raw_eeg = mne.io.read_raw_eeglab(eeg_file)
/tmp/ipykernel_16702/2558830226.py:9: RuntimeWarning: Data will be preloaded. preload=False or a string preload is not supported when the data is stored in the .set file
  raw_eeg = mne.io.read_raw_eeglab(eeg_file)
/tmp/ipykernel_16702/2558830226.py:9: RuntimeWarning: Not setting position of 1 ecg channel found in montage:
['EKG']
Consider setting the channel types to be of EEG/sEEG/ECoG/DBS/fNIRS using inst.set_channel_types before calling inst.set_montage, or omit these channels when creating your montage.
  raw_eeg = mne.io.read_raw_eeglab(eeg_file)

Setting up high-pass filter at 1 Hz

FIR filter parameters

Designing a one-pass, zero-phase, non-causal highpass filter:
- Windowed time-domain design (firwin) method
- Hamming window with 0.0194 passband ripple and 53 dB stopband attenuation
- Lower passband edge: 1.00
- Lower transition bandwidth: 1.00 Hz (-6 dB cutoff frequency: 0.50 Hz)
- Filter length: 1651 samples (3.302 sec)

Setting up high-pass filter at 1 Hz

[... 180 lines cut ...]
   (...)
   2412                 'proj_id', 'proj_name', 'projs', 'sfreq', 'gantry_angle',
   2413                 'subject_info', 'sfreq', 'xplotter_layout', 'proc_history']

ValueError: Measurement infos are inconsistent for dig
```
Although other attributes of prep seem to be fine:

```
prep.EEG_new

array([[ 6.92872951e-19, -2.80886176e-06, -5.02157417e-06, ...,
         5.30101354e-06,  2.26315846e-06,  5.75982404e-19],
       [ 1.00119294e-18, -2.10187357e-06, -2.50465512e-06, ...,
         2.35277111e-07, -3.63729870e-06, -7.86046575e-19],
       [-7.64950161e-19,  1.81967321e-06,  1.10303559e-05, ...,
         1.34994738e-05,  7.49501261e-06, -6.28498447e-19],
       ...,
       [ 4.10578045e-18, -2.07593091e-07,  6.23457305e-08, ...,
         8.15795696e-06,  1.91804891e-06,  1.62630326e-19],
       [-1.05879118e-18, -6.16081013e-07, -7.84905784e-07, ...,
         8.63973602e-06,  1.81714657e-06,  5.14996032e-18],
       [-3.32936888e-18, -1.16585488e-06, -1.58660128e-06, ...,
         9.17628802e-06,  3.79875150e-06,  1.15196481e-18]])
```

sappelhoff, 2022-03-27:
Thanks for the report, this might have to do with the recent MNE-Python 1.0 release ... are you using mne at version 1.0?

Perhaps the warning is spurious for our case and could be supressed.

walkerped, 2022-03-27:
Yes, I am using MNE v. 1.0.0.  It seems to be causing any process I run with prep.raw to fail, rather than just being a warning.  So if I run `EEG = prep.raw.get_data()` and get the error, I'll get a `NameError: name 'EEG' is not defined` if I try to refer to EEG.  Similarly if I try to do `prep.raw.plot()`, I get the `ValueError: Measurement infos are inconsistent for dig` error again.

sappelhoff, 2022-03-28:
please see what I wrote in #114 

Also, please double check your code and see the warnings that your code produces:

> /tmp/ipykernel_16702/2558830226.py:9: RuntimeWarning: Unknown types found, setting as type EEG:
heog: ['HEO']
pupil: ['R-Dia-X-(mm)', 'R-Dia-Y-(mm)']
veog: ['VEO']
  raw_eeg = mne.io.read_raw_eeglab(eeg_file)
/tmp/ipykernel_16702/2558830226.py:9: RuntimeWarning: Data will be preloaded. preload=False or a string preload is not supported when the data is stored in the .set file
  raw_eeg = mne.io.read_raw_eeglab(eeg_file)
/tmp/ipykernel_16702/2558830226.py:9: RuntimeWarning: Not setting position of 1 ecg channel found in montage:
['EKG']
Consider setting the channel types to be of EEG/sEEG/ECoG/DBS/fNIRS using inst.set_channel_types before calling inst.set_montage, or omit these channels when creating your montage.
  raw_eeg = mne.io.read_raw_eeglab(eeg_file)

walkerped, 2022-03-29:
The update in https://github.com/sappelhoff/pyprep/pull/114 seems to have fixed it - I can plot prep.raw, and see that the bad channel was interpolated.  Thanks!

PR #114 (2022-03-29): MNE-Python 1.0 fixes
# PR Description

intended to fix #113 eventually. First by adding some CI steps to catch such issues earlier in the future.
closed 2022-03-29 by walkerped