sappelhoff/pyprep #127: Issue with RawBrainVision

Sam54000, 2022-11-18:
Hi, thank you for this amazing pipeline which works well with mne.
I have an issue though, I am working with mne-bids and my files are converted into BrainVision format. Therefore when I read back my files and try to run PrepPipeline I run into an error
 ```
find_noisy_channels.py", line 92, in __init__
    self.n_samples = raw._data.shape[1]
AttributeError: 'RawBrainVision' object has no attribute '_data'
```
I might find a walkaround by doing `raw._data = raw.get_data()` but I don't think it is a sustainable solution. Someone have any idea?

sappelhoff, 2022-11-18:
Hi, thanks for the report. I am really surprised that the `_data` attribute is not available. I also wonder how mne-bids could have an impact on this. Can you replicate this problem in a minimal working example? Preferably using a file available from mne (e.g., ["sample"](https://mne.tools/dev/overview/datasets_index.html#sample))

sappelhoff, 2022-11-18:
Can you also share some more of the code you use? Also, is your raw object "preloaded" (see https://mne.tools/stable/generated/mne.io.Raw.html) before you pass it to pyprep?

PR #128 (2022-11-20): use get_data instead of ._data

# PR Description

fixes #127 

Not really sure what the problem was over there, but the adjustments here are minor (`get_data()` should yield the same as `._data`, except that it takes a tiny bit longer) and it'd fix the issue.

# Merge Checklist

- [ ] the PR has been reviewed and all comments are resolved
- [ ] all [CI][what-is-ci] checks pass
- [ ] (if applicable): the PR description includes the phrase `closes #<issue-number>` to [automatically close an issue][auto-close-documentation]
- [ ] (if applicable): bug fixes, new features, or [API][what-is-api] cha [... cut]
closed 2022-11-20 by sappelhoff