sappelhoff/pyprep #142: Possible to add an argument in find_bad_by_nan_flat() to change FLAT_THRESHOLD ?

nabilalibou, 2024-02-06:
Hello, 
Kudo for this amazing package, I love reproducibility and automatisation when processing data.
I observed that some nearly flat channels are not flagged neither by_deviation nor by_nan_flat (something that could be resolved by https://github.com/sappelhoff/pyprep/issues/83 ?).

Do you think an argument to toy with in find_bad_by_nan_flat() to change FLAT_THRESHOLD would be interesting ?

sappelhoff, 2024-02-07:
Hi! 

> Do you think an argument to toy with in find_bad_by_nan_flat() to change FLAT_THRESHOLD would be interesting ?

Sure, that looks like something that could be easily added! Do you want to prepare a pull request for it? Else, I can look into it -- let me know.

---

solving: 

- #83 

would require much more thought and experimentation, I am afraid

nabilalibou, 2024-02-07:
ok i'll do a PR, i think it will do more or less the trick before #83 is resolved

PR #144 (2024-02-09): ADD flat_threshold argument in find_bad_by_nan_flat method from NoisyChannels
- PR closes #142
closed 2024-02-09 by sappelhoff