sappelhoff/pyprep #146: add channels in raw.info['bads'] to unusable channels in Reference class

john-veillette, 2024-05-31:
Currently, PREP assumes you have no prior information about which channels are bad. This isn't necessarily a bad design choice if it's philosophically important that all bad channel identification be automatic, but currently it will result in an error if you have any channels marked as bad in `raw.info['bads']` before running PREP.

The culprit re: the error is in `Reference.__init__`:
```
self.raw.pick_types(eeg=True, eog=False, meg=False)
```
`Raw.pick_types` will not include bad channels in its output by default, and since `Reference` inherits `self.prep_params["ref_chs"] = self.ch_names_eeg` from the `PrepPipeline` class that initializes it (where channels manually marked as bad _haven't_ been removed yet), then the `self.raw.get_data(picks=self.reference_channels)` in `Reference.robust_reference` will throw a missing value error.

So a quick solution would be to change the offending line to 
```
self.raw.pick_types(eeg=True, eog=False, meg=False, exclude=[])
```
so that manually marked bad channels are not unintentionally removed. But they should probably also be added to `Reference.unusable_channels`, so users have the ability to manually tell PREP to leave out certain channels (e.g. if they know the channel is broken in their hardware). 

In the meantime, users can circumvent the problem by just setting `raw.info['bads'] = []` before running PREP. Hopefully that saves someone some trouble :)

sappelhoff, 2024-06-01:
> So a quick solution would be to change the offending line to
>
> `self.raw.pick_types(eeg=True, eog=False, meg=False, exclude=[])`
>
> so that manually marked bad channels are not unintentionally removed. [...]

Thanks for the report, I'd be happy to receive a PR for this.

john-veillette, 2024-06-01:
Great, I'll submit a pull request as a soon as I have time!

sappelhoff, 2024-08-29:
@john-veillette any news on your expected timeline for this?

john-veillette, 2024-08-29:
Sorry @sappelhoff this slipped my mind, but I have time to work on it today so this week or next seems reasonable.

john-veillette, 2024-08-29:
Okay, that didn't take long once I got to it. I've now opened PR #156

PR #156 (2024-09-17): Reference interpolates instead of deletes channels manually marked as bad, closes #146

# PR Description

 

Fixes #146, an error that results from the `raw.pick_types` call inside `pyprep.Reference` deleting channels which have been marked as bad prior to PREP. These channels should now be interpolated instead.

A new field `'bad_manual'` is added to `noisy_channels_before_interpolation` dict so there's still a record the channel was interpolated. This happens inside `Reference`; the `NoisyChannels` class is untouched. 

# Merge Checklist

- [ ] the PR has been reviewed and all comments are r [... cut]
commit 89e6fc93 (2024-09-17): Reference interpolates instead of deletes channels manually marked as bad, closes #146 (#156)

* allows manually marked bad channels to be excluded from reference / interpolated, without being removed (which had been causing an error)

* adds new contributor to CITATION.cff

* adds new contributor ORCID

* Update authors.rst

* Update changelog.rst

* fixes some formatting issues flagged by CI

* adds manual bad channels to find_noisy_channels

* changes name of manual bads to match find_noisy_channels (which requires a specific format to work properly)

* adds test coverage

* fixes bug and f [... cut]
closed 2024-09-17 by sappelhoff