sappelhoff/pyprep #151: Move from `pick_types` to `inst.pick`

sappelhoff, 2024-08-14:
See for example: https://github.com/sappelhoff/pyprep/blob/82390acff4c15a8e8ca92ed92c46f7a50d72e049/pyprep/find_noisy_channels.py#L57

This raises:

> NOTE: pick_types() is a legacy function. New code should use inst.pick(...).

Ayush-Devs, 2024-08-20:
Hello, This is my first PR, but I am confident I will be able to update the code. Can you assign this to me? and I will start working on it immediately.

Ayush-Devs, 2024-08-25:
The raw.pick() method returns the modified Raw object itself, not a numpy.ndarray like pick_types() used to.

        if len(arys) == 1:
>           result = asanyarray(arys[0])
E           ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (60,) + inhomogeneous part.

not really sure how to work around this issue.

sappelhoff, 2024-08-25:
Hey @Ayush-Devs, I am making quite a few changes in the background regarding the pyprep infrastructure right now, so it'd be good to not work on this for a while. I had assumed you closing your PR meant that.

sappelhoff, 2024-08-26:
relevant:

- https://github.com/mne-tools/mne-python/issues/11913

closed 2024-11-23 by sappelhoff